pokusew/nfc-pcsc #1: Ability to control LED indicators and buzzer

pokusew, 2016-10-29:
Add support for some devices with this fuction (e.g. ARC122U)

pokusew, 2017-01-15:
Work in progress. Added basic support for ARC122 USB in [`led-control-support`](https://github.com/pokusew/nfc-pcsc/tree/led-control-support) branch.
Note, that APDU commands are different for different devices.

**Plan**
- [ ] wait for resolving pokusew/node-pcsclite#4: Expose vendorId and productId in reader object
- [ ] create new classes for supported readers

indiecore-fm, 2017-02-20:
I'm not finding any ARC122 only ACR122, are they the same device?

pokusew, 2017-02-20:
@indiecore-fm it is ACR122 (ARC122 does not exist, just a typo)

pokusew, 2017-05-28:
Basic support for ACR122 added in https://github.com/pokusew/nfc-pcsc/commit/1477d2314dbbe49ca45587ecdba7efc2c39978c3 and published in [0.5.0](https://github.com/pokusew/nfc-pcsc/releases/tag/v0.5.0)

closed 2017-05-28 by pokusew