pokusew/nfc-pcsc #3: Invalid response length 6. Expected length was 9 bytes.

SuThaw, 2016-11-09:
I got that message when i try to read or write MiFare 1K card. What is wrong ?

pokusew, 2016-11-09:
Hi,

thank you for reporting this bug.
I have looked into and I have noticed that there were expected format of UID when getting card UID.
I have removed this limitation.

Now you should be able to read card UID when you attach it to your NFC reader wtihout any problems.

Can you test it? If everything is OK, I will close this issue.

SuThaw, 2016-11-10:
I still get the following error in using autoprocessing.
[image]

Here is the code. Thanks for your response
https://gist.github.com/SuThaw/688b0e9570c5d72a0074da05d63fe437

pokusew, 2016-11-10:
Hi,

**I am so sorry.** I have mistakenly forgotten to run build before publishing the 0.3.0 version, so the dist folder in 0.3.0 was outdated and still contained the bug.

Now I have properly run build before publishing and **published the new version 0.3.1.** Everything should now work in version 0.3.1 as I have written before.

Could you test it? If everything is OK, I will close this issue.

Thank you very much.

SuThaw, 2016-11-10:
Hi @pokusew 
I can read the uid. Thanks for that. But i still have one more problem. I cannot read the data. It show the following message. I read on block 8 length 16. When I tried to write , I also got the error. Thanks for your changes and response.
[image]

pokusew, 2016-11-11:
Hi @SuThaw,

This error means that the operation (reading/writing) could not be completed successfully  (response status code is 0x9000).

I haven't tested reading Mifare 1K Card because I don't have any (I am using Mifare Ultralight)

But I have looked into some [docs](http://cache.nxp.com/documents/data_sheet/MF1S50YYX_V1.pdf?pspll=1). According to them, it looks like you must authenticate using authenticating keys (A or B, depending which blocks you want to access) before you can read/write data.

Here is the documentation of ARC122U API [API-ACR122U-2.03.pdf](https://github.com/pokusew/nfc-pcsc/files/586266/API-ACR122U-2.03.pdf). There are methods for loading authentication keys and authenticating blocks. You can send them to the reader using [`reader.transmit`](https://github.com/pokusew/nfc-pcsc/blob/master/src/Reader.js#L367) method.

I could integrate this commands to this library, if it is useful for you.

I hope my answer will help you.

btw. you can join here https://discord.gg/bg3yazg and we can solve this issue faster using chat

pokusew, 2016-11-13:
Hi @SuThaw,

I have finally implemented support for authentication. Keys are automatically loaded into reader's memory when authenticating. See https://github.com/pokusew/nfc-pcsc/blob/authentication-support/test/index.js.

I have created new branch [`authentication-support`](https://github.com/pokusew/nfc-pcsc/tree/authentication-support) for testing this feature. Once you confirm it works I will merge it into master and I will release new version and publish it to npm.

Could you clone this branch and test if everything works for you?

Thank you.

SuThaw, 2016-11-13:
@pokusew 
Thanks for your support :) . When i check for authentication, I got the following error.
[image]

pokusew, 2016-11-17:
Hi @SuThaw,

I am sorry for late response, but I have been very busy.

It's strange that you have got this error.

It could be caused by incompatibility of authenticate command with version of your reader. According to the API specification, I have added support for obsolete version of authenticate command too. For easier debugging, I have also added console.log to log content of sent APDU commands packets.

Could you test it if it solves the problem?

Please make sure that you are using correct authentication key and key type for blocks that you want to access.

If the problem persists, we can chat to solve this faster. I believe it should work.

SuThaw, 2016-11-18:
Hi @pokusew 
Thank for your response. I feel sorry for giving trouble to you.  I am still getting the following error. 
[image]

pokusew, 2016-11-18:
Hi @SuThaw,

It is different error than last time.
It looks you have removed the card from the reader too fast but it could be something else.

I am ready for chat.

pokusew, 2017-01-28:
This issue should be resolved as authentication support for Mifare Classic cards has been merged to master and released in [v0.4.0](https://github.com/pokusew/nfc-pcsc/releases/tag/v0.4.0).

closed 2017-01-28 by pokusew