pokusew/nfc-pcsc #5: Windows packaging error

haribabu500, 2017-01-19:
I recently made an electron project with a sample code as mentioned below...

`
//============================= nfc begin============================  
          // 'use strict';

          var _nfcPcsc = require('nfc-pcsc');

          var _nfcPcsc2 = _interopRequireDefault(_nfcPcsc);

          function _interopRequireDefault(obj) { return obj && obj.__esModule ? obj : { default: obj }; }

          var nfc = new _nfcPcsc2.default();

          // var mod=require('nfc-pcsc');
          // var nfc=new mod.NFC();
           
          nfc.on('reader', reader => {
           
              // console.log(`${reader.reader.name}  device attached`);
           
              // needed for reading tags emulated with Android HCE 
              // custom AID, change according to your Android for tag emulation 
              // see https://developer.android.com/guide/topics/connectivity/nfc/hce.html 
              reader.aid = 'F222222222';
           
              reader.on('card', card => {
                  console.log("==================================================");
                  var cardid=card.uid;
                  console.log(cardid);
                  
                  mainWindow.loadURL('http://localhost/AA-EMR/interface/main/card_dashboard.php?card_id='+cardid);
           
              });
           
              reader.on('error', err => {
                  console.log(`${reader.reader.name}  an error occurred`, err);
              });
           
              reader.on('end', () => {
                  console.log(`${reader.reader.name}  device removed`);
              });
           
          });
           
          nfc.on('error', err => {
              console.log('an error occurred', err);
          });  
          //============================= nfc end============================`

Packaging the project for liniux seems to be okay but when I package it for windows, I get the following error...

`Error X A JavaScript error occurred in the main process Uncaught Exception: Error: %1 is not a valid Win32 application. MCAUsers \ Dell \ Desktop \ DMS_Releasa\ DMS-win32-x64 resourca\ app\ nod..Apcsclite.node at Error (native) at process.module.(anonymous function) [as dlopen] (ELECTRON_ASARjs:173:20) at Object.Module„ectensions..node (modulejs:583:18) at Object.module.(anonymous function) [as .node] (ELECTRON_ASARjs:173:20) at Module.load (modulejs:473:32) at tryModuleLoad (modulejs:432:12) at Function.Module,load (modulejs:424:3) at Module.require (modulejs:483:17) at require (internaVmodulejs:20:19) at bindings (C: \ Users\ Dell \ Desktop\ DMS_Releases\ DMS-win32-x64\resources\app modules\bi...:44)`

Any help would be really helpful..
Thank You
Hari

pokusew, 2017-01-21:
Hi @haribabu500 

You are building your Electron app for Windows on Windows machine and you are getting error, right?
What about running your app (without building), does it work?

I am using `nfc-pcsc` in my own Electron project and it builds and runs well on Windows and also on macOS.

Can you send me your code? I could try to build it by myself and find out find out the cause of error.

You can join here https://discord.gg/bg3yazg and we can solve this problem faster using chat.

haribabu500, 2017-01-22:
No sir I am building electron project in linux machine. when I try to package it for windows, I get this error...

pokusew, 2017-01-22:
Hi @haribabu500 

The problem is that you have to build Node native modules (which nfc-pcsc uses) on platform (os and arch), where the app will run.
You can't do cross-build with native modules (to tell the truth, you can, but you must have prebuilt binaries and most libraries don't offer it and it is very complicated).

Also, if you want build your app for Windows, you have to do it on Windows, for macOS, on macOS and so on.

Common solution is to use CI/CD server to build your app. You can connect it to your repo and it will watch branch for new commits and it will build automatically.

For Windows builds, I highly recommend you to use [AppVeyor](https://www.appveyor.com/). It is Continuous Delivery service for Windows and it works pretty well.

Hope, it helps.

closed 2017-01-23 by pokusew