pokusew/nfc-pcsc #11: When using Angular2/TypeScript: TypeError: exists is not a function

philoliver, 2017-03-18:
Hi,
I have an Angular CLI project running in Electron. When I follow your instructions and import the module like this `import NFC from '../../../node_modules/nfc-pcsc';`, and call `const nfc = new NFC(minilogger);`, then I get the following error in the console:

> TypeError: exists is not a function

And during the build process

>WARNING in ./~/bindings/bindings.js
>76:22-40 Critical dependency: the request of a dependency is an expression

philoliver, 2017-03-19:
I did some research and found out that the `TypeError: exists is not a function` appears to be a bug in the `node-bindings` module. See: https://github.com/TooTallNate/node-bindings/pull/20

It should be fixed when the new release is available on NPM.

However, going one step further, this fix will still not make the it work when using AngularCLI/Webpack. To make it work in this scenario we would need to use` node-loader` instead of `node-bindings` to load the binding file. 

I did the following change to make it work locally:
```
"dependencies": {
  "bindings": "github:tootallnate/node-bindings#pull/20/head", < Instead of this,
  "node-loader": "^0.6.0", << use this
  "nan": "^2.5.1"
},
```
Then edited "/node_modules/@pokusew/pcsclite/lib/pcsclite.js"
`const pcsclite = require('node-loader!./../build/Release/pcsclite.node');`

pokusew, 2017-03-21:
Hi @philoliver 

Thank you for posting your problem.

I am quite busy now, but I'll look at it today or tomorrow.

tobinbc, 2017-04-02:
Hi @philoliver @pokusew 
Just to let you know that I have had exactly the same error and this fixed it also. Currently using a gulp task to search + replace the code post install. Thanks @philoliver  for finding the fix!
Cheers

dhruv217, 2017-11-12:
Updating the node-binding to 1.3.0 does not seem to have  solved the issue as @philoliver pointed out. Is there a permanent solution to this problem instead of using gulp because i am a noob and am not familiar with it.
Thanks for the awesome lib by the was. If not for this I would not be able to make my code platform independent. 
I am using an Electron app with angular 4 to read and write data in nfc tags.

pokusew, 2017-11-12:
Hi @dhruv217,

I think it should work. Check out [Can I use this library in my angular-electron app?](https://github.com/pokusew/nfc-pcsc#can-i-use-this-library-in-my-angular-electron-app) in FAQ section of the README. In the linked https://github.com/pokusew/nfc-pcsc/issues/24#issuecomment-327038188, there is described what needs to be done in order to make it work.

It is related to the [angular-electron](https://github.com/maximegris/angular-electron) boilerplate, but I think, that's what you are using in your project https://github.com/dhruv217/shrabik-smart-card-reader.

Please let me know, if it works for you. If you encounter any problems feel free to contact me. 🙂 
I am available for chat here https://discord.gg/bg3yazg.

Hope it helps. 🙂

---

PS Don't forget to star ⭐️  my library, if you find it useful. 😃 Thanks.

Canos, 2019-05-04:
@philoliver thanks for your contribution.
After making your changes proposed above i've got this error

```
Uncaught Error: Cannot open /***/node_modules/@pokusew/pcsclite/build/Release/pcsclite.node: 
TypeError: Cannot read property 'dlopen' of undefined
    at Object.<anonymous> (pcsclite.node:1)
    at Object../node_modules/node-loader/index.js!./node_modules/@pokusew/pcsclite/build/Release/pcsclite.node (vendor.js:72750)
    at __webpack_require__ (bootstrap:78)
    at Object.<anonymous> (pcsclite.js:4)
    at Object../node_modules/@pokusew/pcsclite/lib/pcsclite.js (pcsclite.js:197)
    at __webpack_require__ (bootstrap:78)
    at Object../node_modules/nfc-pcsc/dist/NFC.js (NFC.js:8)
    at __webpack_require__ (bootstrap:78)
    at Object../node_modules/nfc-pcsc/dist/index.js (index.js:145)
    at __webpack_require__ (bootstrap:78)
```

i'm frozen here, dont know how to continue.
i'm using Angular 7.2 and electron 5.0.1

any idea?

naveed-31, 2019-06-12:
I facing the similar issue ,any update on this issue ??

nmattela, 2021-09-02:
+1
I tried philoliver's solution as well and got:

```
./node_modules/@pokusew/pcsclite/build/Release/pcsclite.node (./node_modules/node-loader/dist/cjs.js!./node_modules/@pokusew/pcsclite/build/Release/pcsclite.node)
TypeError: this.getOptions is not a function
```