pokusew/nfc-pcsc #12: NFC is not a constructor

Khasanboy, 2017-04-06:
I am trying to create application using Electron. In the beginning I was getting problem with node Module versions that was expecting 51 but required 53. I installed Electron 1.5.1 and that error is fixed but now I am getting error that says:
# **TypeError: NFC is not a constructor**

I just used your example and error is coming from this line:

const nfc = new NFC();

did I miss something or is it about node native modules ?

Thanks in advance

pokusew, 2017-04-06:
Hi @Khasanboy,

I addressed the problem. It is caused by Babel `export default` transform behaviour (see http://stackoverflow.com/a/33705077).

If you use Babel transpiler (with ES6 modules support) for you app code too, you can import it like `import NFC from 'nfc-pcsc'`.

**Otherwise you must use `const NFC = require('nfc-pcsc').default`.** (add `.default`).

I know, that's not very intuitive.

**Thank you very much for drawing my attention to this problem.** I'll prepare **a solution** (maybe using [babel-plugin-add-module-exports](https://github.com/59naga/babel-plugin-add-module-exports) and add **info to docs**.

---

To this problem:
> I am trying to create application using Electron. In the beginning I was getting problem with node Module versions that was expecting 51 but required 53. I installed Electron 1.5.1 and that error is fixed...

**Node Native modules must be built with the correct version of Node.js, in which they will be used.**

So, if you install this library with npm, it is built with your machine's Node.js version. But then, you are using the library in Electron and Electron uses its own version of Node.js internally.

**So after installing a library with Node Native Module, you must rebuild the source with correct Electron headers.**  
(for example you can run this [node-gyp](https://github.com/nodejs/node-gyp) command: `cd node_modules/nfc-pcsc && node-gyp rebuild --target=[electronVersion] --arch=[arch] --dist-url=https://atom.io/download/atom-shell`)

The usage with Electron is described in README FAQs section [Can I use this library in my Electron app?](https://github.com/pokusew/nfc-pcsc#can-i-use-this-library-in-my-electron-app).  
There is also a link to [Using Native Node Modules](https://electron.atom.io/docs/tutorial/using-native-node-modules/) guide in Electron documentation to fully understand the problematic.

---

Hope it helps.

Khasanboy, 2017-04-08:
Thanks, This fixed the problem

pokusew, 2017-06-02:
Solved, but introduced a breaking change, see [How do I require/import this library?](https://github.com/pokusew/nfc-pcsc#how-do-i-requireimport-this-library) in README.

closed 2017-06-02 by pokusew