pokusew/nfc-pcsc #13: Control Error in LED Example

therealri, 2017-05-16:
Hey @pokusew,

Thanks for this great package!

I am getting this error when trying to run the led.js example on macOS 10.12.4:

```
*[master][~/Desktop/nfc-pcsc]$ node -r babel-register examples/led.js 
New reader detected ACS ACR122U
12:38:01 AM info ACS ACR122U  device attached
Setting AID to F222222222
AID parsed [ 242, 34, 34, 34, 34 ]

trying to connect CONNECT_MODE_DIRECT 3
status { state: 18, atr: <Buffer > }
changes 18
card removed
connected { type: 3, protocol: 3 }
<Buffer ff 00 52 00 00>
transmitting control <Buffer ff 00 52 00 00> 2
{ ControlError: An error occurred while transmitting control.
    at reader.control (/Users/risean/Desktop/nfc-pcsc/src/Reader.js:327:20)
    at /Users/risean/Desktop/nfc-pcsc/node_modules/@pokusew/pcsclite/lib/pcsclite.js:169:11
  name: 'ControlError',
  code: 'failure',
  previous: 
   Error: SCardControl error: Transaction failed.(0x80100016)
       at Error (native) }
```

pokusew, 2017-05-18:
Hi @therealri,

I think that it caused by the special feature of standard PCSC drivers (in all OSs).

**By default standard system PCSC driver doesn't allow to send any commands in direct mode** (i.e. without any card present). You can still send LED control commands when a card is present but it is quite useless.

To allow sending control (i.e. escape) commands in direct mode, you have to set a special value in registry (when you are on Windows) or in Info.plist of driver (on macOS and Linux). [Here](http://tech.springcard.com/2013/scardcontrol-linux-and-java/) is an article to this problematic, but it is quite old.

I was dealing with this problem too (I am also on macOS and btw using ACR122U too 😄).
**So, I figured out the instructions how to enable this feature in macOS (OS X):**

You have to set the value of `ifdDriverOptions`  to `0x0001` (the default value is 0x0000 which disables control commands).  
**Unfortunately, this file is protected by System Integrity Protection (csrutil), so before editing you have to temporary disable it.**

So step by step:
1. disable System Integrity Protection
    1. Boot to Recovery OS by restarting your machine and holding down Command+R shortcut at startup.
    2. Launch Terminal from the Utilities menu.
    3. Enter the following command: `csrutil disable`
    4. Reboot back to normal OS
    5. _(for more info [see Apple docs here](https://developer.apple.com/library/content/documentation/Security/Conceptual/System_Integrity_Protection_Guide/ConfiguringSystemIntegrityProtection/ConfiguringSystemIntegrityProtection.html))_
2. locate driver's **Info.plist** file at `/usr/libexec/SmartCardServices/drivers/ifd-ccid.bundle/Contents/Info.plist` (Finder -> Go To Folder or Command-Shift-G)
3. open **Info.plist** in your text editor
4. set the value of `ifdDriverOptions`  to `0x0001` and save the changes
_(now after rebooting the example and other control commands should work as expected)_
5. enable System Integrity Protection (as in step 1 just use `csrutil enable`)
6. 🏁 You're done. Control commands should work.

Let me know if you have any problems.

Hope it helps.

therealri, 2017-05-27:
Thanks so much @pokusew!

closed 2017-05-27 by therealri

razorxan, 2017-10-07:
any help on how to enable this on win 10 x64?

mich356c, 2018-04-26:
I figured out where to add the DWORD values, but I'm still unable to connect in direct mode. If anyone stumbles upon this and is able to figure it out, please let me know! Here's how to figure out where to add your values:

1. Open the device manager
2. Locate your device
3. Open `Properties` and go to `Details`
4. Find the vlaue for the property `Device Instance Path`
5. Add your new DWORD values in `HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Enum` +  the path you got from step 4 as described in http://tech.springcard.com/2013/scardcontrol-linux-and-java/ at "Working with MS’ CCID driver"
6. Reconnect the device / restart your PC
7. Check if you can connect in DIRECT mode like in https://github.com/pokusew/nfc-pcsc/blob/master/examples/led.js

The error I get is:

```
{ ConnectError: An error occurred while connecting.
    at reader.connect (...\node_modules\nfc-pcsc\dist\Reader.js:217:20)
  name: 'ConnectError',
  code: 'failure',
  previous: Error: SCardConnect error: The smart card has been removed, so that further communication is not possible.
(0x80100069) }
```

Obviously changing port means you have to re-add the values for that entry.

I'm pretty confident this is the right place, but for some reason it just isn't working :/

hakanpinar, 2018-06-25:
Hey,

I'm also stuck in turning off the buzzer of the ACR122 reader.

Anyone who can help me?

carol-braileanu, 2018-11-12:
@mich356c I am having the same issue with an ACR1222L. I tried your solution but it didn't work. Did you manage to find a solution?

mich356c, 2018-11-12:
@carol-braileanu Unfortunately I haven’t

JohnMcLear, 2019-01-14:
@pokusew Any docs for how to do this on linux?  Debby based.

IrakliTabagari, 2019-01-17:
Hi,
Any progress connecting directly on Win 10?

GCBenlloch, 2019-02-25:
Any update on windows 10? I'm getting: 

```
{ ConnectError: An error occurred while connecting.
    at reader.connect (C:\Users\Gonzalo\Desktop\nfc-reader\node_modules\nfc-pcsc\dist\Reader.js:195:25)
  name: 'ConnectError',
  code: 'failure',
  previous: Error: SCardConnect error: Se ha quitado la tarjeta inteligente, por lo que ya no ser� posible la comunicaci�n.
(0x80100069) }
```

Ionut-Milas, 2019-02-28:
@carol-braileanu Any luck ?

Gafuriks, 2019-03-16:
> I figured out where to add the DWORD values, but I'm still unable to connect in direct mode. If anyone stumbles upon this and is able to figure it out, please let me know! Here's how to figure out [...]

No luck here either. The problem (I think) lies in the manufacturer's driver itself. I was not able to disable the buzzer or the LEDs on Windows 10 even when using the provided ACR122UTool which is made by the same people that made the ACR122U reader/writer. Upon disabling the buzzer and applying changes, the reader responds with a "success" message, but the buzzer doesn't actually turn off.

I tried using older driver versions as well to no avail.

I was able to connect and disable the buzzer and manipulate the LEDs on Linux though (Raspberry Pi 3 using Raspbian).

Might be worth checking if this is possible on Windows 7 or even message the manufacturer to see if they have any suggestions. I'll get back with any more info.

carol-braileanu, 2019-05-10:
Any new info on this? :/

doppioandante, 2019-07-17:
Hello, I'm trying to disable the buzzer with the ACR122U on ubuntu.
 I've modified the Info.plist as shown above, but the control operation still fails.

Sending the turn-off ADPU when a card is placed on the reader works but it will beep when the card is detected for the first time. Any idea?

kvasdopil, 2020-03-20:
Looks like it is possible to disable the speaker when using `@pokesuw/pcslite` library directly.
```
const connect = reader => new Promise((resolve, reject) =>
  reader.connect(
    { share_mode: reader.SCARD_SHARE_SHARED },
    (err, protocol) => err ? reject(err) : resolve(protocol)
  )
);
const send = (reader, data, protocol) => new Promise((resolve, reject) =>
  reader.transmit(
    Buffer.from(data),
    40, protocol,
    (err, data) => err ? reject(err) : resolve(data)
  )
);

const pcsc = pcsclite();
pcsc.on('reader', (reader) => {
  reader.on('status', async (status) => {
    // status changed to 'present'
    if (!(reader.state & reader.SCARD_STATE_PRESENT) && (status.state & reader.SCARD_STATE_PRESENT)) {
      try {
        const protocol = await connect(reader);
        const result = await send(reader, [0xFF, 0xCA, 0x00, 0x00, 0x00], protocol); // read id
        await send(reader, [0xFF, 0x00, 0x52, 0x00, 0x00], protocol); // mute speaker
        console.log("card present", result.subarray(0, result.length - 2).toString('hex'));
      } catch (e) {
        console.error('cannot read data', reader.name, e);
      }
    }
  });
});
```

aximut, 2020-08-26:
For me, the control commands do not work at all and a quick fix seems to be sending everything as a transmit command.
Then, the LED and buzzer control works at least while a card is present.
```
this.nfc.on('reader', async reader => {
      reader.control = reader.transmit;
      ...

      reader.on('card', async card => {
        await reader.led(0b01011101, [0x02, 0x01, 0x05, 0x01]);
```

AdamMcCormick, 2020-08-29:
@aximut has the key solution but it's easier than they outline. I extracted the logic of Reader.js (https://github.com/pokusew/nfc-pcsc/blob/master/src/Reader.js#L215) into my own code and called `reader.control` directly. Using protocol 0 and  `0x003136B0` instead of the constant value it works! How on earth did you discover that value?

aximut, 2020-08-29:
> @aximut has the key solution but it's easier than they outline. I extracted the logic of Reader.js (https://github.com/pokusew/nfc-pcsc/blob/master/src/Reader.js#L215) into my own code and called [...]

@AdamMcCormick ACS included it in their card reader tool.

[image]

Nevertheless, they were lazy enough to not update their API docs and made me drain some hours on this.

Meanwhile, I did some other hacky improvements on this, e. g. do a repeated connect attempt in case it fails once, because if you reconnect to the card shortly after disconnecting from direct mode, it will fail undeterministically. I work with a lot of APIs, but this hardware and/or the Windows driver stack is really crap and a pain for developers, I would not want to be the one responsible for it.
I just wanted to make it work with my time management software and now the days pass on getting this "popular" reader to work as it is supposed to.

marcin-chwedczuk, 2021-04-23:
@pokusew workaround no longer works on Big Sur in which the system volume is mounted read only. 

Related: https://apple.stackexchange.com/questions/395508/can-i-mount-the-root-system-filesystem-as-writable-in-big-sur

bra1nDump, 2021-10-06:
Just remove the buzzer :D https://wf.lavatech.top/ave/wogsd-physically-removing-the-buzzer-from-acr122u

igarshep, 2023-10-13:
Pasting this from the forum that I found this issue attached to. In case anyone's still trying to figure out how to do it on OSX.

It took Apple completely dropping the ball with Sonoma in terms of smart card reader support for the solution to this problem to come to light. In turns out, there is a way to completely ignore the CCID driver that comes with OSX and just install you own. Not sure why I never thought about doing that until I was forced to do it. Here are the steps I took to achieve this:

Make sure you have installed both libusb and pkg-config from brew.
1. Download the latest CCID driver from here https://ccid.apdu.fr/files/ and unzip it
2. In terminal, cd to where the libs from libusb are installed (mine was in /opt/homebrew/Cellar/libusb/1.0.26/lib) and run these 2 commands:
- mv libusb-1.0.dylib libusb-1.0.lib
- mv libusb-1.0.0.dylib libusb-1.0.0.lib
3. cd to where you the CCID stuff and run these 2 commands:
- pkg-config --libs libusb-1.0
- pkg-config --cflags libusb-1.0
4. Install the driver using these commands:
- ./MacOSX/configure
- make
- sudo make install

Assuming you've followed along correctly and everything ran without issues, you should now have a new CCID driver installed in /usr/local/libexec/SmartCardServices/drivers/ifd-ccid.bundle. If you open the bundle, you can find the same Info.plist but this one is editable.

Once you have made all your changes, in order for this driver to take effect you have to reboot the system.

igarshep, 2023-12-01:
> I'm on MAC (Sonoma), my SIP is disabled (and Filevault too), impossible to change ifdDriverOptions to 0x0001 (Info.plist), as soon as I save it, the system resets systematically the value to 0x0000.
> 
> I don't understand, any ideas? Thks

Are you using the ifdDriverOptions in /usr/local/libexec/SmartCardServices/drivers/ifd-ccid.bundle or the one in /usr/libexec/SmartCardServices/drivers/ifd-ccid.bundle? The one in usr/libexec is in a read only part of the OS file structure

igarshep, 2023-12-01:
Odd, my change has survived after reboots and OS updates. What are you using to edit the file?

rdluffy, 2024-04-30:
I m having similar issue as you.
On linux, I manage to do it, via python and by modifying the plist
On MAC, there is no possibility to change the plist contained in the read only path. I tried to install libusb, ccid and pcsc-lite (had to recompile it). The problem i have is the driver can not be loaded...
On windows, I don't manage to perform a successfull SCardConnect in pure C. i have tried to edit the registry but it is for the next step i guess

rdluffy, 2024-04-30:
Update, on Windows you need to SCardConnect with no Protocol value (0)