pokusew/nfc-pcsc #14: non-babel version

rynop, 2017-05-24:
Do you have a non-babel version of this module? Node v7.10 has all the es2015 features I need, so I don't to add the babel dependencies.

Or is the best/only way to clone and run your build then include your code that is put into the `dist` dir?

pokusew, 2017-06-02:
Hi @rynop,

this library uses for example **import/export (ES6 modules)** and these are **not supported** in any Node.js version yet. **Otherwise** all needed ES2015 features, that nfc-pcsc lib uses, are supported in 7.10, but **to support all 7.x Node.js versions**, I need to **still transpile** [some things](https://github.com/pokusew/nfc-pcsc/blob/master/.babelrc).

But **you don't need to use Babel in your app** at all. When you **install nfc-pcsc via npm/yarn** and then require it in your code, the transpiled source from dist is used, so you don't need to worry about anything.

As a part of the recent improvements (issues fixes), I have changed the way how to import/require nfc-pcsc, removed unnecessary Babel presets and plugins, dropped support for older Node.js versions and hugely improved examples and README.

These changes are not published yet, but I will release them as a part of the upcoming **0.6.0 release** in a week or two.

Definitely take a look at these FAQs, which I have added to README to better explain the situation:
* [Do I have to use Babel in my app too?](https://github.com/pokusew/nfc-pcsc#do-i-have-to-use-babel-in-my-app-too)
* [How do I require/import this library?](https://github.com/pokusew/nfc-pcsc#how-do-i-requireimport-this-library)

Hope it helps.

PS Don't forget to star ⭐️  my library, if you find it useful. 😃 Thanks.

rynop, 2017-06-07:
Thanks @pokusew I really appreciate the thorough response and readme update.  I look forward to v `0.6.0`.  Closing this now, because as you mention it does not require babel to use.

closed 2017-06-07 by rynop