pokusew/nfc-pcsc #15: emit a card 'off' event

markreid, 2017-05-25:
First, thanks for writing this library! 

I've got a use for it at the moment where it would be useful if the Reader would emit an event when the card is removed. Had a quick look and it'd be a very quick fix; if I submitted a PR would you be interested in merging it in?

Cheers :)

pokusew, 2017-06-02:
Hi @markreid,

thanks for your interest. **It definitely makes sense to add this feature.** 👍

I have been doing some improvements recently, so I have **implemented this feature** too. 😃

**These changes are not published yet**, but I will release them as a part of the **upcoming 0.6.0 release** in a week or two.

Could you **please test it**, if it works for you as expected? (**clone repo** and then `yarn` and `npm run build`, then you can check e.g. `npm run example-basic`. Thank you very much.

Hope it helps.

PS Don't forget to star ⭐️  my library, if you find it useful. 😃 Thanks.

markreid, 2017-06-03:
Hey @pokusew all working from my end; that's basically how i'd implemented it in my fork. 

Thanks :+1:

pokusew, 2017-07-31:
Hi @markreid,

I finally released and published [0.6.0](https://github.com/pokusew/nfc-pcsc/releases/tag/v0.6.0) version with `card.off` event support among the others changes (check out the [release notes](https://github.com/pokusew/nfc-pcsc/releases/tag/v0.6.0)), so I am closing this issue now.

Hope it helps.

closed 2017-07-31 by pokusew