pokusew/nfc-pcsc #16: Mifare Classic 1K: Authentication Error after Multiple Writes

therealri, 2017-05-27:
Hi @pokusew,

Something weird is happening when attempting multiple writes. I have the following simple script that (1) Authenticates Blocks 4-7, Reads Blocks 4-7 and then Writes to Blocks 4-7.

It seems that the cards become unusable after attempting multiple write operations as they refuse to Authenticate. I have also tried this using promises.

I really like this package due to its quality, straightforward installation across platforms and Electron support.

Any suggestions or advice from your side?

`"use strict";

// #############
// Basic example
// - example reading and writing data on from/to card
// - should work well with any compatible PC/SC card reader
// - tested with Mifare Ultralight cards but should work with many others
// - example authentication for Mifare Classic cards
// #############

import winston from 'winston';
import NFC, { TAG_ISO_14443_3, TAG_ISO_14443_4, KEY_TYPE_A, KEY_TYPE_B } from '../src/NFC';
import pretty from './pretty';

// minilogger for debugging
//
// function log() {
// 	console.log(...arguments);
// }
//
// const minilogger = {
// 	log: log,
// 	debug: log,
// 	info: log,
// 	warn: log,
// 	error: log
// };

const nfc = new NFC(); // const nfc = new NFC(minilogger); // optionally you can pass logger to see internal debug logs

let readers = [];

nfc.on('reader', async reader => {

	pretty.info(`device attached`, { reader: reader.name });

	readers.push(reader);

	// needed for reading tags emulated with Android HCE AID
	// see https://developer.android.com/guide/topics/connectivity/nfc/hce.html
	reader.aid = 'F222222222';

	reader.on('card', async card => {

		// standard nfc tags like Mifare
		if (card.type === TAG_ISO_14443_3) {
			// const uid = card.uid;
			pretty.info(`card detected`, { reader: reader.name, card });
		}
		// Android HCE
		else if (card.type === TAG_ISO_14443_4) {
			// process raw Buffer data
			const data = card.data.toString('utf8');
			pretty.info(`card detected`, { reader: reader.name, card: { ...card, data } });
		}
		// not possible, just to be sure
		else {
			pretty.info(`card detected`, { reader: reader.name, card });
		}

		// Notice: reading data from Mifare Classic cards (e.g. Mifare 1K) requires,
		// that the data block must be authenticated first
		// don't forget to fill your keys and types
		// reader.authenticate(blockNumber, keyType, key, obsolete = false)
		// if you are experiencing problems, you can try using obsolete = true which is compatible with PC/SC V2.01
		// uncomment when you need it

		try {
		
			const key = 'FFFFFFFFFFFF';
			const keyType = KEY_TYPE_A;
		
			// we will authenticate block 4, 5, 6, 7 (which we want to read)
			await Promise.all([
				reader.authenticate(4, keyType, key),
				reader.authenticate(5, keyType, key),
				reader.authenticate(6, keyType, key),
				reader.authenticate(7, keyType, key)
			]);
		
			pretty.info(`blocks successfully authenticated`);
		
		} catch (err) {
			pretty.error(`error when authenticating data`, { reader: reader.name, card, err });
			return;
		}

		// example reading 16 bytes assuming containing 16bit integer
		try {

			// reader.read(blockNumber, length, blockSize = 4, packetSize = 16)
			// - blockNumber - memory block number where to start reading
			// - length - how many bytes to read
			// ! Caution! length must be divisible by blockSize

			const data4 = await reader.read(4, 16);
			pretty.info(`[4] data read`);
			pretty.info(`[4] data converted:`, data4.readInt16BE());

			const data5 = await reader.read(5, 16);
			pretty.info(`[5] data read`);
			pretty.info(`[5] data converted:`, data5.readInt16BE());

			const data6 = await reader.read(6, 16);
			pretty.info(`[6] data read`);
			pretty.info(`[6] data converted:`, data6.readInt16BE());

			const data7 = await reader.read(7, 16);
			pretty.info(`[7] data read`);
			pretty.info(`[7] data converted:`, data7.readInt16BE());

		} catch (err) {
			pretty.error(`error when reading data`, { reader: reader.name, card, err });
		}

		// example write 16bit integer
		try {

			// reader.write(blockNumber, data, blockSize = 4)
			// - blockNumber - memory block number where to start writing
			// - data - what to write
			// ! Caution! data.length must be divisible by blockSize

			const data4 = Buffer.allocUnsafe(16);
			data4.writeInt16BE(400);
			await reader.write(4, data4, 16);
			pretty.info(`[4] data written`, { reader: reader.name, card });

			const data5 = Buffer.allocUnsafe(16);
			data5.writeInt16BE(500);
			await reader.write(5, data5, 16);
			pretty.info(`[5] data written`, { reader: reader.name, card });

			const data6 = Buffer.allocUnsafe(16);
			data6.writeInt16BE(600);
			await reader.write(6, data6, 16);
			pretty.info(`[6] data written`, { reader: reader.name, card });

			const data7 = Buffer.allocUnsafe(16);
			data7.writeInt16BE(700);
			await reader.write(7, data7, 16);
			pretty.info(`[7] data written`, { reader: reader.name, card });

		} catch (err) {
			pretty.error(`error when writing data`, { reader: reader.name, card, err });
		}

	});

	reader.on('error', err => {

		pretty.error(`an error occurred`, { reader: reader.name, err });

	});

	reader.on('end', () => {

		pretty.info(`device removed`, { reader: reader.name });

		delete readers[readers.indexOf(reader)];

		console.log(readers);

	});

});

nfc.on('error', err => {

	pretty.error(`an error occurred`, err);

});
`

pokusew, 2017-05-30:
Hi @therealri,

Just to clarify:
Does the script **fail always** or **does it work just for the first time** (meaning that it allows you to write data just once)?

Thanks for your answer.

therealri, 2017-05-30:
Hey @pokusew,

So it works just the first time but thereafter the card becomes unusable.

So yes, it allows me to write data just once using the Exact Script in the Gist and thereafter it's not possible use the card again.

Thanks!

pokusew, 2017-05-30:
And does it work when you restart the script (CTRL-C and run again)?

pokusew, 2017-05-30:
@therealri I have got the idea, from where the problem comes. ❗️❗️❗️

Please answer my previous question and we can go further.

pokusew, 2017-05-30:
@therealri I think that the problem is here:

I am looking into the [technical documentation](http://www.nxp.com/documents/data_sheet/MF1S50YYX_V1.pdf) of Mifare Classic 1K.

Section 8.6 is about Memory organization.
> ... memory is organized in 16 sectors of 4 blocks. One block
contains 16 bytes.

When accessing these blocks with this library, blocks are numbered as follows:
- sector 0:
  - block 0 – manufacturer data (read only)
  - block 1 – data block
  - block 2 – data block
  - **block 3** – sector trailer
- sector 1:
  - block 4 – data block
  - block 5 – data block
  - block 6 – data block
  - **block 7** – sector trailer
- sector 2:
  - block 8 – data block
  - block 9 – data block
  - block 10 – data block
  - **block 11** – sector trailer
- and so on

**That's important, it's that the last block in each sector is "sector trailer" (Section 8.6.3), which contains access conditions for the other 3 data blocks.**

So **block 7**, which you are trying to write to in your code, is not for data. After you write to it, **it changes access conditions (including key)** and you **cannot authenticate** your card then.

I don't have any Mifare Classic card to test it, but according to documentation it is quite clear.

**Could you please update your code (remove interaction with block 7, see updated gist [here](https://gist.github.com/pokusew/a97092ad3e350ad471fbc8347565928c)) and test if it works?**

Hope it helps and looking forward to your answer.

pokusew, 2017-06-02:
Hi @therealri,

I have just noticed another problem in you gist, on line [100](https://gist.github.com/therealri/96adfa625d8bb3b2929a956ec1919cda#file-gistfile1-txt-L100), [104](https://gist.github.com/therealri/96adfa625d8bb3b2929a956ec1919cda#file-gistfile1-txt-L104), [108](https://gist.github.com/therealri/96adfa625d8bb3b2929a956ec1919cda#file-gistfile1-txt-L108), [112](https://gist.github.com/therealri/96adfa625d8bb3b2929a956ec1919cda#file-gistfile1-txt-L112) you forgot to add the third, blockSize, argument 16 (4, 16, **16**). It probably causes unneeded read operations.

Just to be sure, you probably do know it, 16bit integer takes just 2 bytes, not 16 bytes (1 bytes = 8 bits). So before writing only **2-byte** 16-bit integer into **16-byte** unsafely-allocated Buffer, add `data.fill(0)`, more info [here](https://nodejs.org/api/buffer.html#buffer_buffer_from_buffer_alloc_and_buffer_allocunsafe).

I fixed all these things [in my fork here](https://gist.github.com/pokusew/a97092ad3e350ad471fbc8347565928c). Take a look!

Also let me know, if everything works for you as expected.

Hope it helps.

therealri, 2017-06-07:
Hey @pokusew,

Thanks so much and sorry for not getting back to you sooner, I have been travelling.

I also needed to get some extra Mifare Classic 1K Stock.

I have read through your comments and see where I went wrong on the configuration. I appreciate the updated code and all your help and time!

I have tested and it works perfectly, I have a clear understanding of the Mifare Classic 1K Data Layout now :)

Thanks so much!

closed 2017-06-07 by therealri

vamsideepak, 2018-10-08:
it is not working with angular 6 and it throwing error while importing like "nfc is not a constructor". 
import {NFC} from 'nfc-pcsc' throwing error 
ERROR in ./node_modules/bindings/bindings.js
Module not found: Error: Can't resolve 'fs' in 'C:\Users\Admin\Desktop\electronapp\node_modules\bindings'
ERROR in ./node_modules/bindings/bindings.js
Module not found: Error: Can't resolve 'path' in 'C:\Users\Admin\Desktop\electronapp\node_modules\bindings'

reason i found fs module will not work with angular 6. by inbuilt nfc containing fs .  

because of these reason i accessed  nfc from html and i declared in angular 6 component declare var NFC: any; 
because of above i am getting nfc is not a constructor

fabioloreggian, 2019-04-05:
@therealri I have been having troubles trying to read some data from my Mifare Class 1k card. Do you think you could spare some time and help out?

pokusew, 2019-04-05:
Hi @fabioloreggian, what troubles do you have? I am happy to help. 🙂