pokusew/nfc-pcsc #17: DESFire Example Error

therealri, 2017-05-30:
Hey @pokusew,

I've noticed the DESFire example doesn't work with a Blank Card. Is there something I am missing? I am currently looking into this option instead of using Mifare Classic.

Thanks again!

```
C:\Users\User\Desktop\nfc-pcsc-master>npm run example-desfire

> nfc-pcsc@0.5.0 example-desfire C:\Users\User\Desktop\nfc-pcsc-master
> node -r babel-register examples/desfire.js

7:54:34 PM info ACS ACR122 0  device attached
7:54:40 PM info ACS ACR122 0  card detected
card:
  atr:
    0: 59
    1: 129
    2: 128
    3: 1
    4: 128
    5: 128
  standard: TAG_ISO_14443_4
[step 1 - select app] sending <Buffer 90 5a 00 00 03 00 00 00 00>
[step 1 - select app] received data <Buffer 91 00>
[step 2 - authenticate] sending <Buffer 90 0a 00 00 01 00 00>
[step 2 - authenticate] received data <Buffer 6b 58 43 85 e0 ad 5d 80 91 af>
[step 2 - set up RndA] sending <Buffer 90 af 00 00 10 d5 f5 58 dd 18 23 28 47 50 c0 b9 ec 75 ea 8d 6c 00>
[step 2 - set up RndA] received data <Buffer bd 52 c8 3c f3 59 ac eb 91 00>
[step 3 - read] sending <Buffer 90 bd 00 00 07 02 00 00 00 0e 00 00 00>
[step 3 - read] received data <Buffer 91 9d>
7:54:40 PM error ACS ACR122 0  error occurred during processing steps
Error: error in step 3 - read
    at C:/Users/User/Desktop/nfc-pcsc-master/examples/desfire.js:158:11
    at next (native)
    at step (C:\Users\User\Desktop\nfc-pcsc-master\examples\desfire.js:25:191)
    at C:\Users\User\Desktop\nfc-pcsc-master\examples\desfire.js:25:361
```

pokusew, 2017-05-30:
Hi @therealri,

you are right, I haven't noticed it.
**Thank you.**

I prepared the desfire example, when I was helping somebody with debugging issues with implementing DESFire authentication protocol.

Working with a Mifare DESFire card is a bit tricky, because DESFire cards' protocol and technical documentation are under [NDA](https://en.wikipedia.org/wiki/Non-disclosure_agreement) meaning that they are not publicly accessible on the Internet. 😄

It looks, that there is a problem with the read command (which is ok, it would be worse, if there was a problem with authentication 😄).
**Probably the file doesn't exist or something similar.**

We need to take a look into documentation and fix the example according to it.
Currently, I am too busy to do it right now.

Could you take a look at it? I would highly appreciate it.

PS I don't know your usecase, but it is always good to choose card type according to it. Sou if you don't need the complexity of DESFire cards (which are more expensive btw), Mifare Classic could be OK. I will look into it, right now. I am available for chat here https://discord.gg/bg3yazg.

Hope it helps.

jpclair, 2018-02-15:
Hi @pokusew 
In your example, you try to read files stored in the Master app. I'm not sure you can create files on 0x000000 AID (maybe I'm wrong), and not sure that is a good practice to do so :).

I made a PR with the addition of the method GetApplicationIDs
onIDs https://github.com/pokusew/nfc-pcsc/pull/35
It's just a draft, so tell me.

closed 2023-02-18 by riseansmal