pokusew/nfc-pcsc #19: Error while transmitting: Card was reset

SamVerschueren, 2017-06-08:
I often get a `WriteError` but can't really find a workaround for this. I also have no clue what the error message even means. Does someone have a clue how I can solve this?

```js
{ WriteError: An error occurred while transmitting.
    at /Users/sam/Projects/playground/usb/node_modules/nfc-pcsc/dist/Reader.js:573:11
    at throw (native)
    at step (/Users/sam/Projects/playground/usb/node_modules/nfc-pcsc/dist/Reader.js:16:191)
    at /Users/sam/Projects/playground/usb/node_modules/nfc-pcsc/dist/Reader.js:16:402
  name: 'WriteError',
  code: null,
  previous:
   { TransmitError: An error occurred while transmitting.
       at reader.transmit (/Users/sam/Projects/playground/usb/node_modules/nfc-pcsc/dist/Reader.js:271:20)
     name: 'TransmitError',
     code: 'failure',
     previous:
      Error: SCardTransmit error: Card was reset.(0x80100068)
          at Error (native) } }
```

pokusew, 2017-06-08:
Hi @SamVerschueren,

firstly, could you please provide more details?
* What **operating system** do you use?
* What is your **Node.js version**?
* What **card** (which type – Mifare Classic, Ultralight, DESFire, or something else?) are you trying to access?
* What **card reader** do you use?
* **your code** – create a gist/repo or post here
* Does **the error occur by every run**, or is it rather a **random behaviour**?
* Also **make sure**, the card is closely attached to the reader, while running your code. If it is removed/reattached during the writing process, it could lead to similar errors.

According to my quick [search](https://www.google.cz/search?q=0x80100068&oq=0x80100068&aqs=chrome..69i57j69i60l3.204j0j7&sourceid=chrome&ie=UTF-8), it looks like a strange error connected to Windows OS.

Also I have found [an interesting thread](https://social.technet.microsoft.com/Forums/ie/en-US/0bb51532-7fe0-44fb-a493-6e0d7b4c4709/smart-card-reset-problem-new-in-windows-8?forum=w8itprohardware) related to this problem.

I believe, we can find a solution together.

So, please answer my questions.

Hope it helps.

SamVerschueren, 2017-06-09:
Hi @pokusew, thanks for the quick reply. I know, my initial post was very brief but didn't have much time to provide more details, so here we go.

- **Operating System**: macOS Sierra (10.12.4)
- **Node.js version**: 6.9.1
- **Card**: Mifare Ultralight
- **Card reader**: ACR122U

Here is the code that shows the behaviour

```js
'use strict';
const NFC = require('nfc-pcsc').default;

const nfc = new NFC();

nfc.on('reader', reader => {
	reader.on('card', card => {
		console.log(`Card detected (${card.type})`);

		const data = Buffer.from([0xa0, 0xa0, 0xa0, 0xa0, 0xa0, 0xa0, 0xa0, 0xa0, 0xa0, 0xa0, 0xa0, 0xa0]);

		reader.write(6, data)
			.then(result => {
				console.log('done!');
				console.log(result);
			})
			.catch(err => {
				console.log('error');
				console.error(err);
			});
	});
});
```

Save that file as `index.js` and run it. Put a tag on the card reader and then it should fail with the error message that I gave in my initial post. If you kill the program, put the tag on the card reader and then start the program while the tag is already detected by the reader, it works perfectly fine. It just doesn't work if the reader detects the tag while my code is already running.

When I reduce the size of the data buffer to 8 bytes, it seems to work perfectly fine. So I tried to chop my data array in chunks of 4 bytes each and wrote them sequentially to the tag. The first 2 or 3 chunks work, but then I get the same error or another one indicating that the card is unpowered.

So maybe I'm doing something wrong here, but I need to write 64 bytes (or more) to the tag and am not sure how I can achieve that.