pokusew/nfc-pcsc #20: TypeError: NFC is not a constructor

abhilash1in, 2017-07-28:
```
const { NFC } = require('nfc-pcsc');
const nfc = new NFC();
            ^

TypeError: NFC is not a constructor
```

roydejong, 2017-07-28:
Check out https://github.com/pokusew/nfc-pcsc/issues/12

pokusew, 2017-07-31:
Hi @abhilash1in,

you were probably reading docs for master branch (unpublished version), while 0.5.0 was published.

**Nevertheless, I have just published [0.6.0 version](https://github.com/pokusew/nfc-pcsc/releases/tag/v0.6.0)**.  
**Please upgrade nfc-pcsc to 0.6.0 in your project** and it should **work without problems**, when you require it like you did:
```javascript
const { NFC } = require('nfc-pcsc');
const nfc = new NFC();
```

Hope it helps.

abhilash1in, 2017-07-31:
Works great now! Thanks!

closed 2017-07-31 by abhilash1in