pokusew/nfc-pcsc #21: Can not make it run with electron

MajPay, 2017-08-01:
Hello,

thanks for this great package! I already developed some business logic an am able to access my cards via nodejs. Thats fun, unfortunately i can not make it run with electron. Can you please tell me what versions of electron and nfc-pcsc are compatible? I rebuilt the modules like explained but i always get the same kind of error:

```
Error: The module '.../node_modules/@pokusew/pcsclite/build/Release/pcsclite.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 51. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
```

I am simply too stupid to figure it out :-(

changing nodejs version does not help since electron seems to use its own node version.

thanks in advance

MajPay, 2017-08-01:
I got it working now.

1) i used electron 1.7.5 (beta)

2) create a **.npmrc** file in project root, add following content

```
runtime = electron
target = 1.7.5
target_arch = x64
disturl = https://atom.io/download/atom-shell
```

Then: npm rebuild

closed 2017-08-01 by MajPay

francescocatalano, 2018-10-01:
This worked for me (electron 3.0.0):
https://github.com/electron/electron-rebuild
Then =>  ```$(npm bin)/electron-rebuild```