pokusew/nfc-pcsc #22: Reading nfc card data (string) I have written using gotoTags

quocnguyen123, 2017-08-02:
How can I read nfc card data (string) I have written by using gotoTags?

jpclair, 2018-02-25:
Hi, goToTags works with NDEF records, so I think you can.
Please have a look at [this part of the documentation](https://github.com/pokusew/nfc-pcsc/blob/a6cc93c75929d391c162665cbc9c9ab524e6ecf9/README.md#can-i-read-a-ndef-formatted-tag)

somq, 2018-02-25:
Take a look at [this PR](https://github.com/pokusew/nfc-pcsc/pull/31) referencing [this library](https://github.com/somq/nfccard-tool).
> [nfccard-tool - How to parse a ndef message?](https://github.com/somq/nfccard-tool#3---parse-a-ndef-message)

@pokusew Will you take some time to review the PR ?