pokusew/nfc-pcsc #23: can't turn off buzzer

philippschwemberger, 2017-08-31:
Hi,
First of all, thanks for this library. It's really easy to use and works perfectly for my project. 

But I stumbled across a problem while using it in an electron app and tried **turning off the beep signal**.

```
try {
  reader.connect('CONNECT_MODE_DIRECT').then(()=>{console.log('connected')
    reader.setBuzzerOutput(false).then(()=>{console.log('buzzer off')
    reader.disconnect().then(()=>{console.log('disconnected off')}).catch((err)=>{console.error('can\'t disconnect... reason', err)});
    }).catch((err)=>{console.error('can\'t turn off buzz', err)})
  }).catch((err)=>{console.error('can\'t connect', err)})
} catch (err) {
  console.error(err)
}
```
I get this **error message**:
```
┏ Electron -------------------

  can't turn off buzz { Error: An error occurred while transmitting control.
      at reader.control (/home/pi/rfid-play/node_modules/nfc-pcsc/dist/Reader.js:296:20)
      at /home/pi/rfid-play/node_modules/@pokusew/pcsclite/lib/pcsclite.js:169:11
    name: 'ControlError',
    code: 'failure',
    previous: Error: SCardControl error: Transaction failed.(0x80100016) }
  
┗ ----------------------------
```
Do you have an idea how to solve this issue? Or maybe you know a better way how to turn the beep sound off.

Thank you very much.
Philipp

pokusew, 2017-09-01:
Hi @philippschwemberger,

I assume, you are using the ACR122U reader (**am I correct?**).

I think, that it might be **caused by the default behaviour of PC/SC driver**.  
By default you are not allowed to send any (control) commands to the reader (they will fail), when no card is present (e.g. in CONNECT_MODE_DIRECT). However, this might be changed via the driver config.

The [instructions](https://github.com/pokusew/nfc-pcsc/issues/13#issuecomment-302482621), how to change the driver config, can be found in the project's README section Frequent errors – [Transaction failed error when using CONNECT_MODE_DIRECT](https://github.com/pokusew/nfc-pcsc#transaction-failed-error-when-using-connect_mode_direct).

It is written for macOS, but the same (or very similar) applies also for Linux (You wrote, that you are using Raspberry Pi 3).

It is just the idea, I don't know, if I am correct.

If you are using ACR122U, please **let me know** if changing driver config helps. Otherwise please let me know what reader you are using.

Hope it helps. 🙂 

---

Just, to answer this:
> Or maybe you know a better way how to turn the beep sound off.

I think, that's the best way how to do it.
_However (assuming the issue is caused by driver config), if you are unable to change driver config (e.g. you have many devices and changing its driver configs is not applicable), you can workaround this issue and call setBuzzerOutput once the first card is attached (e.g. in CONNECT_MODE_CARD)(naturally it will beep once, but just once)._

---

PS Don't forget to star ⭐️  my library, if you find it useful. 😃 Thanks.

philippschwemberger, 2017-09-01:
Hey, 
Thank you very much for this fast and detailed response :+1: 
Exactly right, I am using a ACR122U reader.

I'll check it during the weekend and let you know if it worked.
cheers

pokusew, 2017-09-09:
Hi @philippschwemberger,

any progress on it? 🙂 

Did you manage to make it work?

There’s no need to hurry. Just don't forget to **let me know** once it works, so I can close the issue then. 🙂 

Thank you very much.

philippschwemberger, 2017-09-11:
Hey @pokusew 

sorry for the delay. But I was a bit stuck with this project. 

I couldnt find the path to the driver, that is specified in the [article](http://tech.springcard.com/2013/scardcontrol-linux-and-java/), you linked in the other issue:
`/usr/local/lib/pcsc/drivers/ccid/Info.plist`
There was no `pcsc` folder in my `lib` folder. I thought that's because i never had installed the ccid-driver, so i installed, CCID [following this article](https://community.nxp.com/docs/DOC-334952) (I installed the other things before, of course, but didnt remember installing CCID seperately) 

But still no folder at `/usr/local/lib`

So I just looked for the `Info.plist` via `sudo find / -name Info.plist` and voila I found the file under `/usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist`

I have to mention, that I tried all of this on my ubuntu 16.04 PC in order to test.

So I changed the file to `0x0001` and ... I couldnt read with the nfc-reader any more :( 
Instead I got this errors messages when I open `systemctl status pcscd.service` (altough the pcscd.servide is active and running) : 
```
sept. 11 12:38:51 chacal systemd[1]: Started PC/SC Smart Card Daemon.
sept. 11 12:38:51 chacal pcscd[5825]: 00000000 hotplug_libudev.c:122:HPReadBundleValues() Cannot open PC/SC drivers directory: /usr/local/lib/pcsc/drivers
sept. 11 12:38:51 chacal pcscd[5825]: 00000008 hotplug_libudev.c:123:HPReadBundleValues() Disabling USB support for pcscd.
```
I set the value of `ifDriverOptons` back to `0x0000` but still I cant read NFC tags anymore on my PC.

But now the **good news**: :)
I thought, maybe I messed up with installing the CCID driver on the PC, so I just gave it a shot on the raspberry pi and looked for a `Info.plist` file.
I found it again under the same path `/usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist` (without installing anything new), simply changed the value of `ifDriverOptions` like described in the article, and with my electron code from above the buzzer is turned off! :smile: 

Thank you very much for your support!

(And if you have any ideas, what could be the cause for the issues on my PC, I would be even more grateful)

cheers,
Philipp