pokusew/nfc-pcsc #24: Problem with Webpack?

drajvver, 2017-09-04:
Hello, I've got empty angular-electron app and wanted to add NFC to it. The problem is that when I copy the first example to HomeComponent I can't compile it.

`ERROR in vendor.bundle.js from UglifyJs
Unexpected token: name (ACR122Reader)` 
error comes out, any ideas?

pokusew, 2017-09-04:
Hi @drajvver, 

I'll sum up (just for the evidence) what we discussed and found out during our [chat](https://discord.gg/bg3yazg):

1.  According to the angular-electron [documentation – Use NodeJS Native libraries](https://github.com/maximegris/angular-electron#use-nodejs-native-libraries):

    > If you need to use NodeJS native libraries, you MUST add it manually in the file webpack.config.js in root folder ...

     So in this case you need to add `"nfc-pcsc": "require('nfc-pcsc')"` in the webpack.config.js externals section (for example below https://github.com/maximegris/angular-electron/blob/master/webpack.config.js#L199).

2. Don't forget that you need to [rebuild the native modules for Electron](https://electron.atom.io/docs/tutorial/using-native-node-modules/).
You can do it for example [this way](https://github.com/drajvver/nfc-pcsc-angular-error/pull/1/commits/c4dbebd9cafd892c90c07194dea0762807b95185#diff-b9cfc7f2cdf78a7f4b91a753d10865a2R23) (just note that's not the only solution). 🙂 

I have created a PR https://github.com/drajvver/nfc-pcsc-angular-error/pull/1 with these changes in your not working example code.

It works for me and I hope you will be able to make it work too (just follow the instructions in the PR).

Hope it helps.

---

PS Don't forget to star ⭐️  my library, if you find it useful. 😃 Thanks.

closed 2017-09-09 by pokusew

Canos, 2019-05-04:
There is no webpack.config.js in recent versions of angular (as of 6.0).  
So how can we configure nfc-pcsc in angular +6?

commit 981674db (2017-09-09): Add note about usage in angular-electron (close #24)