pokusew/nfc-pcsc #25: Write operation failed: Status code: 0x6300

bdsomer, 2017-09-09:
A very simple program that just writes to a card:
```javascript
const { NFC } = require('nfc-pcsc');
const nfc = new NFC();

nfc.on('reader', (reader) => {
	reader.on('card', async (card) => {
		try {
			const data = Buffer.allocUnsafe(16);
			data.fill(0);
			const randomNumber = Math.round(Math.random() * 1000);
			data.writeInt16BE(randomNumber);

			await reader.write(4, data);
		} catch (err) {
			console.error(err)
		}
	})
})
```

When a card is touched to the reader, the error occurs:
```
{ Error: Write operation failed: Status code: 0x6300
    at /.../node_modules/nfc-pcsc/dist/Reader.js:578:11
    at Generator.next (<anonymous>)
    at step (/.../node_modules/nfc-pcsc/dist/Reader.js:18:191)
    at /.../node_modules/nfc-pcsc/dist/Reader.js:18:361
    at <anonymous> name: 'WriteError', code: 'operation_failed' }
```

What's causing this? Is this a bug?

I was getting the same status code at a lower level, as well.

pokusew, 2017-09-09:
Hi @bdsomer,

it is certainly not a bug (hopefully). 😄 

**Firstly, what card you you use? (Mifare Classic?)**

In case, that you use Mifare Classic, you need to **authenticate every block, that you want to access** (write to). That you do with the following:
```javascript
await reader.authenticate(4, KEY_TYPE_A, 'ffffffffffff');
```
_Note: `reader.loadAuthenticationKey` is internal API, you don't need to use it. It is called automatically and in highly efficient manner, when you call `reader.authenticate`._

Further, if you use Mifare Classic, you need to **pass the third parameter (blockSize)** to `reader.write`. Mifare Classic cards have **block size 16**. So you can write data like this:
```javascript
await reader.write(4, data, 16);
```

Please also be ware of the **memory organization (structure) of Mifare Classic** card – [see my comment](https://github.com/pokusew/nfc-pcsc/issues/16#issuecomment-304989178) explaining it.
 _You should to be cautious, when writing more than 16 bytes of data with single `reader.write` call. For example if you want to write 64 bytes of data starting in block number 4 using block size 16 (needed for Mifare Classic) and you call `reader.write(4, data, 16)`, under the hood it will result in four commands (`reader.write(4, data1, 16)`, `reader.write(5, data2, 16)`, `reader.write(6, data3, 16), reader.write(7, data4, 16)`) (e.g. 64 / 16 = 4). The problem is the memory organization of Mifare Classic card. Last block in each sector is "sector trailer" (in this case 7) which contains access conditions for the other 3 data blocks (for 4, 5, 6) in the sector._

Finally, take a look at the [main example](https://github.com/pokusew/nfc-pcsc/blob/master/examples/index.js), where everything is explained in comments (note that you need to uncomment some code to make it work with Mifare Classic). Just read the commets I you will be okay. 🙂 

That should be all. Please let me know if it solves your problems.

Hope it helps.

bdsomer, 2017-09-09:
@pokusew Awesome, thanks! This solved my problem.

To fix, I followed your steps:
- Remove `await reader.loadAuthenticationKey(0, 'ffffffffffff')`
- Pass the third parameter to the `reader#read` and `reader#write` methods

Thanks, again. 👍

closed 2017-09-09 by bdsomer

Philip-Nunoo, 2017-09-13:
@bdsomer could you give a full descripton and better summary. I'm facing similar issues here as well.

thanks.

pokusew, 2017-09-13:
Hi @Philip-Nunoo,

firstly, I suppose you use **Mifare Classic** card, right?

I think that everything is summed up in my previous comment https://github.com/pokusew/nfc-pcsc/issues/25#issuecomment-328284926. There is a link to [the example](https://github.com/pokusew/nfc-pcsc/blob/master/examples/index.js), where everything is explained with comments and code (you need to uncomment some code for Mifare Classic as there is described) 🙂 

Just a small **recapitulation** – how to read/write data on **Mifare Classic** card _(just for your convenience, but be sure to read my previous comment, links and example)_:

1. You need to **authenticate every block, you want to access** (read or write), using `reader.authenticate` as shown [here](https://github.com/pokusew/nfc-pcsc/blob/master/examples/index.js#L77) (remove Promise.all if you don't need to authenticate more blocks). Ensure, that you use the correct key (default is `ffffffffffff `, as far as I know) and the correct key type.

2. Then don't forget to pass **the third parameter** – block size **16** (number) to the `reader.read` and `reader.write` calls.

3. Lastly be aware of the **memory organization (structure) of Mifare Classic** card. See my [previous comment](https://github.com/pokusew/nfc-pcsc/issues/25#issuecomment-328284926).

Hope it helps. 🙂 

---

PS Don't forget to star ⭐️  my library, if you find it useful. 😃 Thanks.

Philip-Nunoo, 2017-09-14:
@pokusew nice. Was able to figure it out and thanks for the prompt reply as well. 

I'd sure :star: it ayt.