pokusew/nfc-pcsc #27: "Failed to minify code" (create-react-app)

haakonmt, 2017-09-21:
I'm trying to use this library together with resin.io's [electron + react template](https://github.com/resin-io-playground/resin-electronjs-react), which uses create-react-app.

When I run the build script it fails telling me that this library (specifically `dist/Reader.js`) can't be minified and directs me to [this link](http://bit.ly/2tRViJ9), which suggests filing an issue on the dependency in question, so here I am.

Native node modules + create-react-app + electron is asking for trouble, I know, but maybe you have some suggestions?

pokusew, 2017-09-22:
Hi @haakonmt,

thank you for posting your issue.

I personally use nfc-pcsc inside an **Electron app which is built using Webpack** and it **works great**, but it requires some config (in some cases): 
1. if you use **Webpack** (create-react-app uses it), you need to add `nfc-pcsc` to `externals` section in webpack.config.js _(it is because it cannot be loaded and built with Webpack as it loads Node native modules)_
  
    ``` javascript
    {
      // ... your Webpack object
      externals: ['nfc-pcsc']
    }
    ```
2. be aware of Node Native modules usage in Electron, see FAQ [Can I use this library in my Electron app?](https://github.com/pokusew/nfc-pcsc#can-i-use-this-library-in-my-electron-app) in README (it might work out of box, but in most cases you need to rebuild modules)

I hope you will be able to make it work with these info. 🙂 

However, if you get stuck, you can create repo with your code (it can be just a minimal example), where the issue occurs and I'll take a look at it (clone, install, run).

Hope it helps.

haakonmt, 2017-09-22:
Thank you for the fast response!

Adding the library to webpack's externals seems like a perfectly good solution, but I forgot to add that I would try to avoid ejecting if possible.

Nevertheless, I found a solution! By using `window.require` to import the nfc module (`const { NFC } = window.require('nfc-pcsc');`), I was able to make webpack ignore it altogether, while still making it work in a node environment (in my case electron).

closed 2017-09-22 by haakonmt

Ionut-Milas, 2018-07-27:
@haakonmt How did you managed to get this running with React without eject ?
Where did you required the files ?

haakonmt, 2018-07-27:
@Ionut-Milas Wherever you want! For instance in `src/nfc.js`:

```
const { NFC } = window.require('nfc-pcsc');

const nfc = new NFC();

nfc.on('reader', reader => {
});
```

Then at the top of `src/index.js`: `import './nfc';`

Ionut-Milas, 2018-07-27:
Thanks a lot !