pokusew/nfc-pcsc #29: Got àè&!é!!&"é echo'd on example/index

notflip, 2017-11-08:
Any idea what this can mean? I don't get any message saying 'card detected..'.
Thank you.

pokusew, 2017-11-08:
Hi @notflip,

thank you for posting your issue. Could you please provide more details so that we can find out where the problem is? 🙂 

* What **operating system** do you use?
* What is your **Node.js version**?
* **How** do you run the example?
    * As described in [Running examples locally](https://github.com/pokusew/nfc-pcsc#running-examples-locally)?
_If not, please try to do it so._
    * **What do you see** when you run the example? (`npm run example`).
_You should see a message saying "device attached" when a reader is available._
   * Can you try to **run the basic example** (`npm run example-basic`)?
_You should see a message saying "device attached" when a reader is available._

Please answer the questions so we can go on. Thanks.

Hope it helps.

pokusew, 2017-11-25:
@notflip Any updates? Were you able to make it work?
Please let me know so that we can eventually close the issue. 🙂

notflip, 2017-11-25:
Sorry my bad it's a faulty reader. :) your repo is just fine!

pokusew, 2017-11-25:
@notflip OK thank you for your prompt reply. I closed the issue.

closed 2017-11-25 by pokusew