pokusew/nfc-pcsc #30: Write NDEF record

nicolasbadia, 2017-12-11:
Hi,
First thanks for this library!
I cannot manage to write an NDEF record to a card. I believe it should be possible...
Here is what I tried:

   
    const { NFC } = require('nfc-pcsc');
    const nfc = new NFC();
    const ndef = require('@taptrack/ndef');

    nfc.on('reader', reader => {
      reader.aid = 'F222222222';

      reader.on('card', async card => {
        try {
          var textRecord = ndef.Utils.createUriRecord("http://www.exemple.com");
          var message = new ndef.Message([textRecord]);
          var bytes = message.toByteArray();

          await reader.write(4, bytes);
        } catch (err) {
          console.error(`error while writing data`, err);
        }
      });
    });

Which gave me this error:

```
error while writing data TypeError: "list" argument must be an Array of Buffers
    at Function.Buffer.concat (buffer.js:332:13)
    at /home/pi/node_modules/nfc-pcsc/dist/Reader.js:562:26
    at Generator.next (<anonymous>)
    at step (/home/pi/node_modules/nfc-pcsc/dist/Reader.js:18:191)
    at /home/pi/node_modules/nfc-pcsc/dist/Reader.js:18:437
    at /home/pi/node_modules/nfc-pcsc/dist/Reader.js:18:99
    at Reader.write (/home/pi/node_modules/nfc-pcsc/dist/Reader.js:583:5)
    at /home/pi/node_modules/nfc-pcsc/dist/Reader.js:546:27
    at Generator.next (<anonymous>)
    at step (/home/pi/node_modules/nfc-pcsc/dist/Reader.js:18:191)
```

Am I missing something?

pokusew, 2017-12-11:
Hi @nicolasbadia,

I think that the problem is here:

The `data` argument of `reader.write(blockNumber, data)` **must be an instance of Node.js [Buffer](https://nodejs.org/api/buffer.html)**, but `message.toByteArray()` returns ES [Uint8Array](https://developer.mozilla.org/cs/docs/Web/JavaScript/Reference/Global_Objects/Uint8Array) (according to their [documentation](https://www.npmjs.com/package/@taptrack/ndef#uri-records)).

So, you need to **convert the Uint8Array into to the Buffer**. You can do it **like this**: 
```javascript
Buffer.from(arr.buffer) // in your code it is Buffer.from(bytes.buffer)
```
**That's the most efficient way** (the one which shares the memory between Uint8Array and Buffer instances). You can read more about the differences between Buffers and TypedArrays [here](https://nodejs.org/api/buffer.html#buffer_buffers_and_typedarray).

Here is the fixed code:
```javascript
const { NFC } = require('nfc-pcsc');
const nfc = new NFC();
const ndef = require('@taptrack/ndef');

nfc.on('reader', reader => {
  reader.aid = 'F222222222';

  reader.on('card', async card => {
    try {
      const textRecord = ndef.Utils.createUriRecord("http://www.exemple.com");
      const message = new ndef.Message([textRecord]);
      const bytes = message.toByteArray();
      // convert the Uint8Array into to the Buffer 
      const data = Buffer.from(bytes.buffer);

      // data is instance of Buffer
      await reader.write(4, data);
    } catch (err) {
      console.error(`error while writing data`, err);
    }
  });
});
```

Hope it helps. 🙂

pokusew, 2017-12-11:
Hi @nicolasbadia,

I found out another important fact about writing NDEF data on card.

In order to create NDEF formatted card/tag, **you have to encapsulate the NDEF message into TLV blocks** _(more info [here](https://stackoverflow.com/questions/37220910/how-to-interpret-ndef-content-on-mifare-classic-1k) and in the links mentioned there)_.

TLV structure is **needed for the tag to be recognized as NDEF formatted tag**. It allows the applications to read NDEF messages from tags **without knowing the data length** _(I am preparing implementation of `reader.readNdef` and `reader.writeNdef` to this library in order to make it easier to work with NDEF)_.

Let's get back to writing NDEF data. I implemented the encapsulation process.
Here is the **updated** code _(assuming you are using Mifare Ultralight)_:
```javascript
const { NFC } = require('nfc-pcsc');
const nfc = new NFC();
const ndef = require('@taptrack/ndef');

const encapsulate = (data, blockSize = 4) => {

  if (data.length > 0xfffe) {
    throw new Error('Maximal NDEF message size exceeded.');
  }

  const prefix = Buffer.allocUnsafe(data.length > 0xfe ? 4 : 2);
  prefix[0] = 0x03; // NDEF type
  if (data.length > 0xfe) {
    prefix[1] = 0xff;
    prefix.writeInt16BE(data.length, 2);
  } else {
    prefix[1] = data.length;
  }

  const suffix = Buffer.from([0xfe]);

  const totalLength = prefix.length + data.length + suffix.length;
  const excessLength = totalLength % blockSize;
  const rightPadding = excessLength > 0 ? blockSize - excessLength : 0;
  const newLength = totalLength + rightPadding;
	
  return Buffer.concat([prefix, data, suffix], newLength);

};

nfc.on('reader', reader => {
  reader.aid = 'F222222222';

  reader.on('card', async card => {
    try {
      const textRecord = ndef.Utils.createUriRecord("http://www.exemple.com");
      const message = new ndef.Message([textRecord]);
      const bytes = message.toByteArray();
      // convert the Uint8Array into to the Buffer and encapsulate it
      const data = encapsulate(Buffer.from(bytes.buffer));

      // data is instance of Buffer containing encapsulated NDEF message
      await reader.write(4, data);
    } catch (err) {
      console.error(`error while writing data`, err);
    }
  });
});
```

I have testes it and it works! 🚀 The tags/cards written with this code are being correctly recognized by Android.

As I have written above, the encapsulation process will be integrated into nfc-pcsc (as `reader.writeNdef`).

Please let me know if everything works for you. 🙂 

Hope it helps. 🙂

nicolasbadia, 2017-12-12:
Thanks a lot for the help and the quick answer.
I managed to write NDEF records to the card with your code.

I do not have an Android device yet to test if it works, but I will have one soon...

I tried to read the NDEF records with the library, but couldn't manage to do it.
Not sure where the issues come from... but this is another issue...

chelinho139, 2017-12-14:
@nicolasbadia Hey Nico, can you share a reading ndef example? thanks

chelinho139, 2017-12-14:
@pokusew do you have a reading example?

nicolasbadia, 2017-12-17:
@chelinho139 Hi, take a look at our discussion on [discord](https://discord.gg/bg3yazg). I think you will find what you need there.

SuneethaYamani, 2019-09-19:
How to find application id

alifardoss, 2019-11-26:
Whats the update on this .readNDEF and .writeNDEF