pokusew/nfc-pcsc #37: Recovering from Error

bierlair, 2018-03-10:
I'm trying to write some methods that are checking if a card (in my case NTAG-21x) needs authentication or not.

So far what I'm doing is reading 4 Bytes from Page 4 and if it fails, it means that I need to authenticate.

The problem occurs that the error thrown during that reading failure is preventing me from doing anything meaningful until I disconnect and reconnect the card.

Here some code:

```
try {
	let data = await this.reader.read(4, 4);
	return true;
} catch (err) {
	console.log("Cannot read, so we must authenticate!");
}

try {
	// if first try block is commented out, this block here works just fine!!!
	console.log("This fails, even though this area is not password protected");
	let data2 = await this.reader.read(0, 4);
} catch (err) {
	console.log("I will always be thrown, no matter what... WHY?", err);
}

```

Is there a way to reset the state without disconnecting the card?
... or is my authentication approach simply inefficient?

pokusew, 2018-03-13:
Hi @bierlair,

Sorry for the late response. I've been quite busy.

I'll take a look at it this afternoon. I'll test it with my card.

I'll let you know what I figure out. 🙂

bierlair, 2018-03-13:
From what I've found out so far is that once you receive a 0x6300 message, you cannot do anything else until the card reader goes back into "IDLE" mode (usually within 1-2 seconds, when the LED switches back to steady green).

If I was to build some kind of artificial timeout of 2 seconds before both try/catch blocks, it would work fine.

This leads me to believe that it's a by-design feature that cannot be changed... unless there is a certain APDU that forces to go back into IDLE state right away.