pokusew/nfc-pcsc #41: Won't write/read to card - tried to wipe using GoToTags

iamyourabu, 2018-05-29:
Hey there!

I recently found this library and have been trying to get the example to read/write to my card but it keeps erroring. The errors aren't really help as you can see:

```
> nfc-pcsc@0.6.2 example C:\Users\Omar\Documents\Programming\fullstack\nfc-pcsc
> node -r babel-register examples/index.js

08:56:42 info ACS ACR122 0  device attached
(node:2944) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
08:56:44 info ACS ACR122 0  card detected
card:
  atr:
    0:  59
    1:  143
    2:  128
    3:  1
    4:  128
    5:  79
    6:  12
    7:  160
    8:  0
    9:  0
    10: 3
    11: 6
    12: 3
    13: 0
    14: 1
    15: 0
    16: 0
    17: 0
    18: 0
    19: 106
  standard: TAG_ISO_14443_3
  type:     TAG_ISO_14443_3
  uid:      ca318459
08:56:44 error ACS ACR122 0  error when reading data
err:
  name: ReadError
  code: operation_failed
08:56:44 error ACS ACR122 0  error when writing data
err:
  name: WriteError
  code: operation_failed
08:56:53 error ACS ACR122 0  an error occurred
err:
08:56:53 info ACS ACR122 0  device removed
[ <1 empty item> ]
08:59:49 info ACS ACR122 0  device attached
08:59:59 info ACS ACR122 0  card detected
card:
  atr:
    0:  59
    1:  143
    2:  128
    3:  1
    4:  128
    5:  79
    6:  12
    7:  160
    8:  0
    9:  0
    10: 3
    11: 6
    12: 3
    13: 0
    14: 1
    15: 0
    16: 0
    17: 0
    18: 0
    19: 106
  standard: TAG_ISO_14443_3
  type:     TAG_ISO_14443_3
  uid:      ca318459
08:59:59 error ACS ACR122 0  error when reading data
err:
  name: ReadError
  code: operation_failed
08:59:59 error ACS ACR122 0  error when writing data
err:
  name: WriteError
  code: operation_failed
```

I did join your Discord but I have yet to see you on. Would appreciate some assistance.
Thanks!

pokusew, 2018-05-29:
Hi @iamyourabu,

thank you for posting your issue here. 🙂 

**I've noticed that you are trying to run the [main example](https://github.com/pokusew/nfc-pcsc/blob/master/examples/index.js) using `npm run example`, right?**

That example works like this:
1. it waits for a card
2. after detecting a card, it tries to read 16 bytes starting with page 4, assuming page size is 4 bytes (it works well for tags like Mifare Ultralight, but it won't work for all), the read data are printed
3. then it tries to write a random number and overwrites the old data

I think, you are using a different tag/card or your tag/card is not writable (e.g. locked). So you have to make some changes in the code in order to read you tag/card (change page address, page size, data length, ...).

In order we can identify the problem more specifically, could you please provide more details?
* What **card** (which type – Mifare Classic, Ultralight, DESFire, NTAG or something else?) are you trying to read/write? – that's the most important
* What **operating system** do you use?  – assuming *Windows*
* What is your **Node.js version**?
* What **card reader** do you use?  – assuming *ACS ACR122*

---

❗️ **Check out my answer on Discord, as well. According to the infomation you provided, you don't need to read/write cards, you can use just card uids.**

btw _Sorry for my late reply. I am extremely busy now and I must have somehow missed the notification. Usually if someone mentions me, or send me a DM, I'll go online._

So if you need more help, feel free to contact me. 🙂

---

PS Don't forget to star ⭐️ my library, if you find it useful. 😃 Thanks.

iamyourabu, 2018-05-30:
Hey @pokusew!

Thanks for responding. Yes, I ran the example to test out the library. In response to your questions:

1. I used a program called GoToTags and apparently the card is a Mifare Classic 1k.
2. Windows 10 for now, will be deploying my final project on a Raspberry Pi though :)
3. My node version is v9.10.1 and my npm version is the latest 5.6.0
4. You are right, I am using the ACS ACR122 👍 

I looked at your Discord response, and you are spot on with what I am trying to do. I can just the UID of the cards, but the only reason I don't want to is because on my backend, students have IDs and I'd rather have those on my cards instead. I can just end up using IDs, but just want to try and see if I can figure this out.

Would appreciate the help!
Thanks

kkornwinkel, 2018-09-08:
Push Please. Have the Same error. What was the Solution for this?

pokusew, 2018-09-08:
Hi @kkornwinkel,

are you trying to to run the *[main example](https://github.com/pokusew/nfc-pcsc/blob/master/examples/index.js)* using `npm run example` or `yarn run example`?

**What card** (which type – Mifare Classic, Ultralight, DESFire, NTAG or something else?) are you trying to read/write?

**In case, you use Mifare Classic**, you have to authenticate to do read/write and you have to pass the blockSize argument to the `reader.write` – **you should uncomment appropriate lines in the main example** https://github.com/pokusew/nfc-pcsc/blob/master/examples/index.js#L63.

Hope it helps. 🙂 

If you encounter any problems, feel free to drop me a line. 😉