pokusew/nfc-pcsc #48: Starting issue with electron

IonutBurlacu, 2018-10-13:
Hello. I'm developing a React+Electron app and I'm also using this library to read some bands with a nfc reader. The problem I have is that on some computers the app runs perfectly, but on the others, the app starts as a background process and it stays there with no error.
Any idea what can cause this to happen?
[image]

danielhanzs, 2018-10-17:
Attach a card-reader to the computer then start the program, it issome logic problem that if no card reader found it freezes the program withput any error response.

IonutBurlacu, 2018-10-17:
@danielhanzs I tried what you said. I installed the needed usb driver and attached the usb reader and now the app works. The strange thing is that it works now even without the nfc reader connected. Like it installed something when it connected and now it doesn't need it anymore. Any idea how to solve this?

IonutBurlacu, 2018-10-17:
I have another person which tried only installing the nfc usb driver, but no success. So it must be related to the usb reader itself.

danielhanzs, 2018-10-17:
If you connect the device and start the application and desconnect the devie without restarting the application it will start searching for the readers instead of stopping the application which is good.

I will try tomorrow the https://www.npmjs.com/package/usb-detection repo

var usbDetect = require('usb-detection');
usbDetect.startMonitoring();
 
usbDetect.on('add', function(device) {
    console.log(device);
});

On add event I will check if the device, vendorId or productId or deviceName or manufacturer is the one I'm searching for (needed only for ACR122 USB reader) and then I will call the new NFC() class.

I'm at home will test it tomorrow at workplace if physical reader needed to be connected or the installed driver will solve the problem.

For a solid solution I will stick with usb detect.

IonutBurlacu, 2018-10-18:
Ok. Thank you very much!

TeodorDre, 2019-02-26:
> If you connect the device and start the application and desconnect the devie without restarting the application it will start searching for the readers instead of stopping the application which is good.
> 
> I will try tomorrow the https://www.npmjs.com/package/usb-detection repo
> 
> var usbDetect = require('usb-detection');
> usbDetect.startMonitoring();
> 
> usbDetect.on('add', function(device) {
> console.log(device);
> });
> 
> On add event I will check if the device, vendorId or productId or deviceName or manufacturer is the one I'm searching for (needed only for ACR122 USB reader) and then I will call the new NFC() class.
> 
> I'm at home will test it tomorrow at workplace if physical reader needed to be connected or the installed driver will solve the problem.
> 
> For a solid solution I will stick with usb detect.
> 
> PS.: For usb-detection you need full vs2017 community the windows-build-tools not sufficient enough to build it.

Hello! it is actually now, does it was succeed?  on 50% of windows computer, i have a problem with it

CoWayger, 2023-07-21:
I don't think this is the "right" solution, there are many card readers on market and manually listing them in some enum is not very maintainable approach. Is there some flag for "not waiting for card reader connection"?

liamcmitchell, 2025-09-24:
I can reproduce this by disabling the SCardSrv service...

[image]

...running my electron app and then clicking "No" in the permission prompt:

[image]

The program prints `Shell Execute failed with 1223` to the console and then freezes.

I think it's this block of code and the freeze is being stuck in the loop on line 72:

https://github.com/pokusew/node-pcsclite/blob/8100373ef05278b274250695577584fb161dec9b/src/pcsclite.cpp#L49-L78

robincsamuel, 2025-10-28:
I'm having the same issue. The app becomes unresponsive on Windows when calling `new NFC()`. It works fine if I comment it out. 

@pokusew , it would be great if you could share your thoughts.