pokusew/nfc-pcsc #50: Unexpected token

emond07, 2018-11-07:
Hi,

Im having problem with the sample code.
Im using node v11

C:\Program Files\nodejs\node.exe --inspect-brk=31374 nfc-pcsc\examples\basic.js
Debugger listening on ws://127.0.0.1:31374/44f7c3dd-ad54-42e3-90ad-014997bf8ae2
For help, see: https://nodejs.org/en/docs/inspector
Debugger attached.
c:\Users\emond\Desktop\nfc\nfc-pcsc\examples\basic.js:8
import { NFC } from "../src/index.js";
^

SyntaxError: Unexpected token {
at new Script (vm.js:80:7)
at createScript (vm.js:264:10)
at Object.runInThisContext (vm.js:316:10)
at Module._compile (internal/modules/cjs/loader.js:670:28)
at Object.Module._extensions..js (internal/modules/cjs/loader.js:718:10)
at Module.load (internal/modules/cjs/loader.js:605:32)
at tryModuleLoad (internal/modules/cjs/loader.js:544:12)
at Function.Module._load (internal/modules/cjs/loader.js:536:3)
at Function.Module.runMain (internal/modules/cjs/loader.js:760:12)
at startup (internal/bootstrap/node.js:308:19)
Waiting for the debugger to disconnect...
SyntaxError: Unexpected token {
vm.js:80
at new Script (vm.js:80:7)
at createScript (vm.js:264:10)
at Object.runInThisContext (vm.js:316:10)

pokusew, 2018-11-10:
Hi @emond07,

thank you for posting your issue. 🙂 

According to the error output you posted, you are trying to run the [basic example](https://github.com/pokusew/nfc-pcsc/blob/master/examples/basic.js), right?

However, the examples use **import/export modules syntax** which is not supported in Node.js by default _(you need some transpiler, e.g. Babel with appopripate plugins, or you can enable native support (with some limitations) in the newest Node.js versions)_.

**The simplest way to run the example is with `npm run example-basic` or `yarn example-basic`. As you can see [here](https://github.com/pokusew/nfc-pcsc/blob/master/package.json#L40) in the package.json, it is actually an alias for `node -r babel-register examples/basic.js`.** Requiring babel-register using -r flag, ensures the code is transpiled applying rules from project's root [.babelrc](https://github.com/pokusew/nfc-pcsc/blob/master/.babelrc).

Hope it helps.

Feel free to contact me, if you encounter any problems or need any help.

---

PS Don't forget to star ⭐️ my library, if you find it useful. 😃 Thanks.

pokusew, 2018-12-28:
Hi @emond07,

any updates on this issue?
Were you able to make it work?

Please let me know so I can close the issue. Thanks. 🙂

pokusew, 2019-01-02:
@emond07 I am closing the issue now. Feel free to reopen it in case you are still facing the issue. 😉

closed 2019-01-02 by pokusew

emond07, 2019-01-05:
hi, 

Im having issues in running the basic.js using NPM

C:\Users\emond\Desktop\nfc\nfc-pcsc>npm run example-basic

> nfc-pcsc@0.7.0 example-basic C:\Users\emond\Desktop\nfc\nfc-pcsc
> node -r @babel/register examples/basic.js

internal/modules/cjs/loader.js:589
    throw err;
    ^

Error: Cannot find module '@babel/register'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:587:15)
    at Function.Module._load (internal/modules/cjs/loader.js:513:25)
    at Module.require (internal/modules/cjs/loader.js:643:17)
    at Module._preloadModules (internal/modules/cjs/loader.js:828:12)
    at preloadModules (internal/bootstrap/node.js:733:7)
    at startup (internal/bootstrap/node.js:298:9)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:878:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! nfc-pcsc@0.7.0 example-basic: `node -r @babel/register examples/basic.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the nfc-pcsc@0.7.0 example-basic script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\emond\AppData\Roaming\npm-cache\_logs\2019-01-05T00_31_28_240Z-debug.log

C:\Users\emond\Desktop\nfc\nfc-pcsc>

pokusew, 2019-01-06:
Hi @emond07,

it seems that you **had not installed the dependencies** before you ran the example.

**Try to install the dependencies with `npm install` first and then run `npm run example-basic`.**

For more info, see [Running examples locally](https://github.com/pokusew/nfc-pcsc#running-examples-locally) section of the README.

Hope it helps. 🙂 

---

PS Don't forget to star ⭐️my library, if you find it useful. 😃Thanks.

emond07, 2019-01-07:
it now working.. thanks for the help

pokusew, 2019-01-07:
@emond07 You're welcome! 🙂
Enjoy using the nfc-pcsc library and don't forget to star ⭐️it, if you find it useful. 😃Thanks.

tk7589, 2019-10-02:
I get this error

SyntaxError: Unexpected token <
    at new Script (vm.js:80:7)
    at createScript (vm.js:274:10)
    at Object.runInThisContext (vm.js:326:10)
    at Module._compile (internal/modules/cjs/loader.js:664:28)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:754:12)
    at startup (internal/bootstrap/node.js:283:19)