pokusew/nfc-pcsc #53: Uncaught Error: DLL Initialization routine failed

mikeneyens, 2018-11-19:
I recently switched computers and re-installed node, electron, etc. Now I'm getting the following error when running my NFC application:

> Uncaught Error: a dynamic link library (dll) initialization routine failed.
\\?\C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\Release\pcsclite.node
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620)
    at tryModuleLoad (internal/modules/cjs/loader.js:559)
    at Function.Module._load (internal/modules/cjs/loader.js:551)
    at Module.require (internal/modules/cjs/loader.js:658)
    at require (internal/modules/cjs/helpers.js:20)
    at bindings (C:\Users\Mike\desktop\pos\node_modules\bindings\bindings.js:81)
    at Object.<anonymous> (C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\lib\pcsclite.js:4)

- Electron version: 3.0.9
Can someone help me?
Thanks in advance!

Mike

pokusew, 2018-11-19:
Hi @mikeneyens,

it seems that **you need to rebuild the native modules.** nfc-pcsc uses Node Native Modules under the hood to access system's PC/SC API. Native modules are built against specfic Node.js runtime (

This library uses pcsclite native bindings pokusew/node-pcsclite under the hood.

Hope it helps. 🙂 

---

PS Don't forget to star ⭐️  my library, if you find it useful. 😃 Thanks.

pokusew, 2018-11-19:
Hi @mikeneyens,

thanks for posting your issue here.

It seems that **you need to rebuild the native modules.** nfc-pcsc uses *Node Native Modules* under the hood to access system's PC/SC API. Native modules are built against **a specific Node.js version present in the runtime** (e.g. against the current Electron.js version which contains a specific Node.js runtime).

After you upgrade Electron, you usually need to rebuild the modules. (_Native modules are automatically built when installing the dependencies._)

**Please see [Using Node native modules in the official Electron guide](https://electronjs.org/docs/tutorial/using-native-node-modules)**.
**I recommend using [electron-rebuild](https://github.com/electron/electron-rebuild).**

You can also read the [Can I use this library in my Electron app?](https://github.com/pokusew/nfc-pcsc#can-i-use-this-library-in-my-electron-app) FAQ in the nfc-pcsc's, where you'll find additional useful information.

FYI: On top of that, native modules depend on the underlaying OS (Windows vs Linux vs macOS), which means they need to be built specifically for each OS and architecture (32bit vs 64bit). This is relevant when targeting more platforms, read more [in the FAQs](https://github.com/pokusew/nfc-pcsc#can-i-use-this-library-in-my-electron-app).

Hope it helps. 🙂 

---

PS Don't forget to star ⭐️  my library, if you find it useful. 😃 Thanks.

mikeneyens, 2018-11-20:
Thanks for the fast response!

Unfortunately I tried rebuilding using electron-rebuild, but I'm getting multiple errors. I opened an issue over there (https://github.com/electron/electron-rebuild/issues/263 ).

mikeneyens, 2018-11-20:
Okay, so I made a stupid mistake: electron was not installed as a devDependency. Electron-rebuild now works. However, when I try to open the app nothing works like it should, even Developer Tools.
When I comment out the portion of NFC code, everything works fine.

I tried re-installing nfc-pcsc, now I'm getting following errors: 

>   c:\users\mike\.node-gyp\10.13.0\include\node\v8.h(2658): note: see declaration of 'v8::String::WriteUtf8' (compiling source file ..\src\pcsclite.cpp)
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\common.h(34): warning C4477: '_snprintf' : format string '%s' requires an argument of type 'char *', but variadic argument 2 has type 'LPVOID' (compiling source file ..\src\ca
rdreader.cpp) [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\common.h(34): warning C4477: '_snprintf' : format string '%s' requires an argument of type 'char *', but variadic argument 2 has type 'LPVOID' (compiling source file ..\src\pc
sclite.cpp) [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\cardreader.cpp(91): warning C4996: 'v8::String::Utf8Value::Utf8Value': was declared deprecated [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj
]
  c:\users\mike\.node-gyp\10.13.0\include\node\v8.h(2891): note: see declaration of 'v8::String::Utf8Value::Utf8Value'
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\cardreader.cpp(94): warning C4996: 'v8::Value::ToString': was declared deprecated [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  c:\users\mike\.node-gyp\10.13.0\include\node\v8.h(10040): note: see declaration of 'v8::Value::ToString'
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\pcsclite.cpp(183): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\cardreader.cpp(137): warning C4996: 'v8::Value::Uint32Value': was declared deprecated [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  c:\users\mike\desktop\pos\node_modules\nan\nan.h(1655): note: see declaration of 'Nan::Callback::Call'
  c:\users\mike\.node-gyp\10.13.0\include\node\v8.h(2477): note: see declaration of 'v8::Value::Uint32Value'
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\pcsclite.cpp(186): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\cardreader.cpp(138): warning C4996: 'v8::Value::Uint32Value': was declared deprecated [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  c:\users\mike\desktop\pos\node_modules\nan\nan.h(1655): note: see declaration of 'Nan::Callback::Call'
  c:\users\mike\.node-gyp\10.13.0\include\node\v8.h(2477): note: see declaration of 'v8::Value::Uint32Value'
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\cardreader.cpp(172): warning C4996: 'v8::Value::Uint32Value': was declared deprecated [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  c:\users\mike\.node-gyp\10.13.0\include\node\v8.h(2477): note: see declaration of 'v8::Value::Uint32Value'
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\cardreader.cpp(218): warning C4996: 'v8::Value::ToObject': was declared deprecated [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  c:\users\mike\.node-gyp\10.13.0\include\node\v8.h(10046): note: see declaration of 'v8::Value::ToObject'
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\cardreader.cpp(219): warning C4996: 'v8::Value::Uint32Value': was declared deprecated [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  c:\users\mike\.node-gyp\10.13.0\include\node\v8.h(2477): note: see declaration of 'v8::Value::Uint32Value'
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\cardreader.cpp(220): warning C4996: 'v8::Value::Uint32Value': was declared deprecated [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  c:\users\mike\.node-gyp\10.13.0\include\node\v8.h(2477): note: see declaration of 'v8::Value::Uint32Value'
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\cardreader.cpp(273): warning C4996: 'v8::Value::ToObject': was declared deprecated [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  c:\users\mike\.node-gyp\10.13.0\include\node\v8.h(10046): note: see declaration of 'v8::Value::ToObject'
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\cardreader.cpp(274): warning C4996: 'v8::Value::Uint32Value': was declared deprecated [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  c:\users\mike\.node-gyp\10.13.0\include\node\v8.h(2477): note: see declaration of 'v8::Value::Uint32Value'
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\cardreader.cpp(275): warning C4996: 'v8::Value::ToObject': was declared deprecated [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
[... cut ...] cxproj]
  c:\users\mike\desktop\pos\node_modules\nan\nan.h(1655): note: see declaration of 'Nan::Callback::Call'
c:\users\mike\desktop\pos\node_modules\@pokusew\pcsclite\src\cardreader.cpp(665): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\Mike\desktop\pos\node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  c:\users\mike\desktop\pos\node_modules\nan\nan.h(1655): note: see declaration of 'Nan::Callback::Call'

I also can't find a prior issue regarding these errors.

pokusew, 2018-12-28:
Hi @mikeneyens,

I am very sorry for the late reply. I've been very busy recently.

It's great that you made electron-rebuild work! 👍 Please close the issue https://github.com/electron/electron-rebuild/issues/263 with an appropriate comment.

It turned out that nfc-pcsc did not support Node.js 10.x (the one Eelectron uses) and 11.x.

**I upgraded all the dependencies and also made a lot of fixes. Support for all recent Node.js versions along with many improvements is available in [v0.7.0](https://github.com/pokusew/nfc-pcsc/releases/tag/v0.7.0).** 🚀 

Please upgrade nfc-pcsc in your dependencies and let me know if it works for you. 🙂

Hope it helps. 🙂 

---

PS Don't forget to star ⭐️my library, if you find it useful. 😃Thanks.

pokusew, 2018-12-31:
Hi @mikeneyens, any update on the issue?
Have you tried to upgrade nfc-pcsc to [v0.7.0](https://github.com/pokusew/nfc-pcsc/releases/tag/v0.7.0) and rebuild it using electron-rebuild? Any success?
Please let me know so I can eventually close the issue. Thanks: 🙂

mikeneyens, 2019-01-02:
Hi Martin

I updated to v0.7.0 and used rebuild, the issue is fixed now. Thanks for the help!

closed 2019-01-02 by mikeneyens