pokusew/nfc-pcsc #56: Can not use ACS ACR1252 1S CL Reader

prabowomurti, 2018-12-06:
I am trying to run the example with `npm run example`. We are using ACR1252U USB NFC Reader, but when I do plug and tap in a card, the reader can't read with the following errors : 

```
> nfc-pcsc@0.6.2 example .../belajar-electron/nfc-reader
> node -r babel-register examples/index.js

2:41:21 PM info ACS ACR1252 1S CL Reader(1)  device attached
2:41:21 PM info ACS ACR1252 1S CL Reader(2)  device attached
2:41:22 PM error ACS ACR1252 1S CL Reader(2)  an error occurred
err: 
2:43:51 PM info ACS ACR1252 1S CL Reader(1)  device removed
[ <1 empty item>,
  Reader {
    _events: { card: [Function], error: [Function], end: [Function] },
    _eventsCount: 3,
    _maxListeners: undefined,
    reader:
     CardReader {
       name: 'ACS ACR1252 1S CL Reader(2)',
       connected: true,
       _events: [Object],
       _eventsCount: 4,
       state: 6 },
    logger:
     { log: [Function: log],
       debug: [Function: debug],
       info: [Function: info],
       warn: [Function: warn],
       error: [Function: error] },
    connection: { type: 2, protocol: 2 },
    card:
     { atr:
        <Buffer 3b df 18 ff 81 f1 fe 43 00 3f 03 83 4d 49 46 41 52 45 20 50 6c 75 73 20 53 41 4d 3b>,
       standard: 'TAG_ISO_14443_4',
       type: 'TAG_ISO_14443_4' },
    autoProcessing: true,
    _aid: 'F222222222',
    _parsedAid: [ 242, 34, 34, 34, 34 ],
    keyStorage: { '0': null, '1': null },
    pendingLoadAuthenticationKey: {} } ]
2:43:51 PM info ACS ACR1252 1S CL Reader(2)  device removed
[ <2 empty items> ]
2:49:20 PM info ACS ACR1252 1S CL Reader(2)  device attached
2:49:21 PM error ACS ACR1252 1S CL Reader(2)  an error occurred
err: 
2:49:26 PM error ACS ACR1252 1S CL Reader(1)  an error occurred
err: 
  name:     GetUIDError
  code:     null
  previous: 
    name:     TransmitError
    code:     failure
    previous: 
```

Here is some logs when I attached the ARC12U NFC Reader : 
```
2:52:47 PM info ACS ACR122U  device attached
2:52:58 PM error ACS ACR122U  an error occurred
err: 
```

I use node v10.14.0 and npm 6.4.1. Any clue or advice is appreciated. Thank you.

Codelica, 2019-03-11:
@prabowomurti did you ever find a solution?  We are looking at using this reader also, and am curious if it's working now.  Thanks.

prabowomurti, 2019-03-12:
@Codelica Hi James. Unfortunately we left the mistery unsolved and continue our life with other web application. Thank you.

alba-ado, 2022-01-20:
This library works fine with the ACR122U reader but fails to authenticate the blocks using the same code on the ACR1252U card reader. Rest of the functionality works ok, but authentication does not work.