pokusew/nfc-pcsc #59: MIFARE Ultralight C authentication

Franx0, 2018-12-24:
First of all, great work, is amazing and super usefull having resources like yours. Thank you!

I'm on firts step on authentication of MIFARE Ultraligh C tags. Related to the doc I need to use a command to have the default tag key in return.
https://www.nxp.com/docs/en/data-sheet/MF0ICU2.pdf (page 15)

I'm using a block like that:
```javascript
reader.on('card', async card => {
	console.log(`${reader.reader.name}  card ${card.uid} inserted`, card);
	try {
		const buffer = Buffer.from([0x1a, 0x00]);
		return resolve(await reader.transmit(buffer, 26));				
	} cache (err) {
                 console.log(err);
        };			
});

The main issue here is that always return a byteArray like that: Uint8Array(2) [99, 0]
instead of the expected 8bits string.
Maybe I'm missing something but the following code: 0x1a, 0x00 is neccessary as first authentication step.

Thank you all ;)

Franx0, 2018-12-30:
***EDIT***

Sending the following buffered command 
```javascript
Buffer.from([0xff, 0x00, 0x00, 0x00, 0x16, 0xd4, 0x42, 0x1a, 0x00])
```
I'm receiving the following:
```
d543019000
```

I'm missing something, becasuse as I read, "01" byte represent the response status as Timeout but I can't understand why.
Any help will be appreciated, thank you all! ;)

pokusew, 2018-12-31:
Hi @Franx0,

I am very sorry for the late reply. Thank you for posting your issue here. 🙂

I faced the similar issues when I implemented MIFARE Ultralight EV1 password authentication (PWD_AUTH).

**I think, the 5th byte (`0x16`, the `number of bytes to send`) should be `0x04` as you are trying to send these 4 bytes: ` 0xd4 0x42 0x1a 0x00`.**

The following should work:
```js
Buffer.from([0xff, 0x00, 0x00, 0x00, 0x16, 0xd4, 0x42, 0x1a, 0x00])
```

**Explanation (might be useful for others):**
(also see the [MIFARE Ultralight EV1 example](https://github.com/pokusew/nfc-pcsc/blob/master/examples/mifare-ultralight-ntag.js#L85) in nfc-pcsc exmap)

As yo figured out correctly, that the only way to send the command, seems to be via the ACR reader's **Direct Transmit** command. It's structure is as follows:

```js
const directTrasmit = Buffer.from([
    0xff,
    0x00,
    0x00,
    0x00,
    data.length, // number of bytes to send
    ...data, // data to send (assuming a Buffer or an array of bytes)
]);
```
The data sent via Direct Transmit are passed directly to the internal NFC frontend chip (in case of ACR readers, that's NXP PN533 or similar). To communicate with the NFC card you have to use the **InCommunicateThru** as described in PN533 API docs. It's structure is as follows:
```js
const inCommunicateThru = Buffer.from([
    0xd4, // Data Exchange Command:
    0x42, // InCommunicateThru
    ...apdu, // data which will be sent directly to the card
]);
```

**So the final structure for sending any direct command to the card for ACR readers might look like this:**
```js
const cmd = Buffer.from([/* the bytes of your command */]);
Buffer.from([
    0xff, // Class
    0x00, // Direct Transmit (see ACR122U docs) (INS)
    0x00, // ... (P2)
    0x00, // ... (P1)
    0x02 + cmd.length, // Lc: Length of the Direct Transmit Payload
    // Payload (2 + cmd.length) bytes
    0xd4, // Data Exchange Command (see PN533 docs)
    0x42, // InCommunicateThru
    ...cmd,
]);
```
The response should like:
```
// d5 43 00 ... ... 90 00
// byte 0: d5 prefix for response of Data Exchange Command (see PN533 docs)
// byte 1: 43 prefix for response of Data Exchange Command (see PN533 docs)
// byte 2: Data Exchange Command Status – 0x00 is success (see PN533 docs, Table 15. Error code list)
// bytes 3-n: Data Exchange Command Response – response from the card
// last 2 bytes: ACR reader status code (should be always success – 90 00)
```

---

**Please let me know if your able to make your command work.**

Also I think, you might be interested in the [MIFARE DESFire example](https://github.com/pokusew/nfc-pcsc/blob/master/examples/mifare-desfire.js), where the 3DES authentication is implemented. It might be similar to the 3DES authentication procedure on the Ultralight C.

Finally, I think that the MIFARE Ultralight C authentication would be great to have in nfc-pcsc examples. Once you finish it, feel free to open a PR to add it. Thanks. 😉

Hope it helps. 🙂 

---

PS Don't forget to star ⭐️my library, if you find it useful. 😃Thanks.

Franx0, 2019-01-02:
> Hi @Franx0,
>
> I am very sorry for the late reply. Thank you for posting your issue here.
>
> I faced the similar issues when I implemented MIFARE Ultralight EV1 password authentication (PWD_AUTH).
>
> **I [...]

Hi @pokusew thanks for your reply. So many usefull doc in that comment ;), thank's a lot. Unfortunately I'm still not able to make the pcsc command works. Let me show you the two command responses:

1. Sending the following:
```javascript
    let buffer = Buffer.from([0xff, 0x00, 0x00, 0x00, 0x04, 0xd4, 0x42, 0x1a, 0x00]);
    reader.transmit(buffer, buffer.byteLength);
```
In that case the response is the tag uid, not the 8byte token.

2. Sending this one:
```javascript
    let buffer = Buffer.from([0xff, 0x00, 0x00, 0x00, 0x16, 0xd4, 0x42, 0x1a, 0x00]);
    reader.transmit(buffer, buffer.byteLength);
```
The response is "d543019000" as my previous comment. I can not figure out what I'm missing or what can be wrong. I debugged deep in order to see nothing weird in the response except this 01 byte that I assume it's an internal code from reader telling me that might exists a timeout issue.

Franx0, 2019-01-02:
EDIT:
After multiple tests and having in mind @pokusew advices, I'm able to tell all of you that @pokusew command were right, but I was having and error in the data length send in the transmit command, so in order to send the command you need something like this:

```javascript
    let buffer = Buffer.from([0xff, 0x00, 0x00, 0x00, 0x04, 0xd4, 0x42, 0x1a, 0x00]);
    return reader.transmit(buffer, 16);
```

Thank's a lot @pokusew, I'll keep on working in the auth, let you know ;)

pokusew, 2019-01-03:
Hi @Franx0,

I've just noticed **a very dangerous mistake** in your previous code snippets. Therefore, I'd like to clarify the usage of `reader.transmit()` method. It might be useful for other people who come across this, as well. 🙂

The [`reader.transmit(data, responseMaxLength)`](https://github.com/pokusew/nfc-pcsc/blob/master/src/Reader.js#L290) method accepts two arguments:
1. data – a Buffer instance containing the data (bytes) to transmit
2. responseMaxLength – an integer specifying the maximum length (in bytes) of the expected response  
_In case you specify too small value and the response exceeds the given maximum length, **the transmit fails and an error is thrown**. If you specify much bigger value than needed, nothing happens – but don't do that for performance reasons._

So, this is the  ☠️**INCORRECT** usage:

```js
// DO NOT USE THIS CODE ☠️
let buffer = Buffer.from([0xff, 0x00, 0x00, 0x00, 0x04, 0xd4, 0x42, 0x1a, 0x00]);
reader.transmit(buffer, buffer.byteLength); // <- this is INCORRECT usage ☠️
```

And this is the ✅**CORRECT** usage:
```js
let buffer = Buffer.from([0xff, 0x00, 0x00, 0x00, 0x04, 0xd4, 0x42, 0x1a, 0x00]);
reader.transmit(buffer, responseMaxLength); // <- fill appropriate expected length of the response
```

In your case (the first part of AUTHENTICATE command), the successful *response* should look like this:
```
d5 43 00 ... ... ... ... ... ... ... ... 90 00
```

- _byte 0:_ `0xd5` prefix for response of Data Exchange Command (see PN533 docs)
- _byte 1:_ `0x43` prefix for response of Data Exchange Command (see PN533 docs)
- _byte 2:_ `0x00` Data Exchange Command Status – 0x00 is success (see PN533 docs, Table 15. Error code list)
- _byte 3-n:_. Data Exchange Command Response – response from the card – in your case, **the 8bytes _ek(RndB)_**
- _last 2 bytes:_ ACR reader status code (should be always success – `0x9000`)
- **Total: 3 + 8 + 2 bytes = 13 bytes** – that's the value you should use as the _responseMaxLength_ in `reader.transmit`

In practice, when developing an implementation of some commands, **I recommend trying the bigger value for _responseMaxLength_ as first, then checking the layout and the length of the received response and finally if the real length matches the calculated value use it as  _responseMaxLength_.**

Hope it helps. 🙂 

---

PS Don't forget to star ⭐️my library, if you find it useful. 😃Thanks.

Franx0, 2019-01-04:
> Hi @Franx0,
> 
> I've just noticed **a very dangerous mistake** in your previous code snippets. Therefore, I'd like to clarify the usage of `reader.transmit()` method. It might be useful for other people who come across this, as well. 
> 
> The [`reader.transmit(data, responseMaxLength)`](https://github.com/pokusew/nfc-pcsc/blob/master/src/Reader.js#L290) method accepts two arguments:
> 
> 1. data – a Buffer instance containing the data (bytes) to transmit
> 2. responseMaxLength – an integer specifying the maximum length (in bytes) of the expected response
>    _In case you specify too small value and the response exceeds the given maximum length, **the transmit fails and an error is thrown**. If you specify much bigger value than needed, nothing happens – but don't do that for performance reasons._
> 
> So, this is the  **INCORRECT** usage:
> 
> ```js
> // DO NOT USE THIS CODE ☠️
> let buffer = Buffer.from([0xff, 0x00, 0x00, 0x00, 0x04, 0xd4, 0x42, 0x1a, 0x00]);
> reader.transmit(buffer, buffer.byteLength); // <- this is INCORRECT usage ☠️
> ```
> 
> And this is the **CORRECT** usage:
> 
> ```js
> let buffer = Buffer.from([0xff, 0x00, 0x00, 0x00, 0x04, 0xd4, 0x42, 0x1a, 0x00]);
> reader.transmit(buffer, responseMaxLength); // <- fill appropriate expected length of the response
> ```
> 
> In your case (the first part of AUTHENTICATE command), the successful _response_ should look like this:
> 
> ```
> d5 43 00 ... ... ... ... ... ... ... ... 90 00
> ```
> * _byte 0:_ `0xd5` prefix for response of Data Exchange Command (see PN533 docs)
> * _byte 1:_ `0x43` prefix for response of Data Exchange Command (see PN533 docs)
> * _byte 2:_ `0x00` Data Exchange Command Status – 0x00 is success (see PN533 docs, Table 15. Error code list)
> * _byte 3-n:_. Data Exchange Command Response – response from the card – in your case, **the 8bytes _ek(RndB)_**
> * _last 2 bytes:_ ACR reader status code (should be always success – `0x9000`)
> * **Total: 3 + 8 + 2 bytes = 13 bytes** – that's the value you should use as the _responseMaxLength_ in `reader.transmit`
> 
> In practice, when developing an implementation of any commands, **I recommend trying the bigger value for _responseMaxLength_ as first, then checking the layout and the length of the received response and finally if the real length matches the calculated value, use the _calculated and verified-is-correct length_ as the final _responseMaxLength_.**
> 
> Hope it helps. 
> 
> PS Don't forget to star my library, if you find it useful. Thanks.

Thank's a lot @pokusew, you're right. The code I posted was wrong. In my case was 16 bytes of response length but it was wrong too so I changed to 13 as you suggested. You're right, the better approach is to check that response length match with expected in order to control the correct transmit flow. 
Thank you a lot ;)

carol-braileanu, 2019-09-25:
Wow, thank you @pokusew, you saved me a lot of time on this :)

dongmai, 2024-12-20:
Hi @pokusew,

Could you please help me with 3DES authentication on my UWP app?

I’m using direct transmit, as per your suggestion above, to perform the authentication. At step 1, everything seems fine—I successfully get the EK(RndB). However, when I send back the EK(RndA || RndB'), I always receive D5-43-02-90-00. I’m quite confident that the logic for 3DES encryption/decryption is correct.

Could you provide any advice on this? Do I need to call any specific command to wake up or activate the card before starting the authentication process?

Below is my function based on your guide above:

```
public byte[] BuildDirectTransmitApdu(byte[] cmd)
{
    // Validate the input command
    if (cmd == null || cmd.Length == 0)
    {
        throw new ArgumentException("Command (cmd) must not be null or empty.");
    }

    // Build the APDU command
    List<byte> apdu = new List<byte>
    {
        0xFF,       // Class byte (CLA)
        0x00,       // Direct Transmit instruction (INS)
        0x00,       // Parameter 1 (P1)
        0x00,       // Parameter 2 (P2)
        (byte)(2 + cmd.Length), // Lc: Length of the payload (2 + command length)
        0xD4,       // Data Exchange Command (from PN533 docs)
        0x42        // InCommunicateThru command
    };

    // Add the payload (raw command)
    apdu.AddRange(cmd);

    // Return the constructed APDU
    return apdu.ToArray();
}

```

Thank you so much for your help!

P.S. I’ve already starred your library.

pokusew, 2024-12-23:
Hi @dongmai,

Thank you for posting your issue here. I'm happy to help you with the 3DES authentication. 🙂

Your `BuildDirectTransmitApdu()` function looks good! 👍 Furthermore, since you say that you are successfully getting the `ek(RndB)`, I don't think there is any problem with that function.

To find out where the problem might be, **I implemented the whole MIFARE Ultralight C 3DES authentication myself** over the weekend. 😅 I have some MIFARE Ultralight C tags (and an ARC122U NFC USB reader), so I was able to iterate and try different things until I got it working. 🚀

**Here is the result:** 👉 [examples/**mifare-ultralight-c.js**](https://github.com/pokusew/nfc-pcsc/blob/master/examples/mifare-ultralight-c.js). It is fully functional and works correctly!  ✅ Moreover, it is thoroughly documented and includes detailed explanations. Below, I'll go over some of the key parts you might find useful.
<details>
<summary>🧑‍💻 <em>You can quickly run the mifare-ultralight-c.js example locally and with the following commands (click to expand) to see if it works for you (it works for me on macOS with ACR122U USB NFC reader):</em></summary>

```bash
git clone https://github.com/pokusew/nfc-pcsc.git
cd nfc-pcsc
npm install
npm run example-mifare-ultralight-c
```

</details>

**Most importantly, based on my testing, I have a few ideas where you might have a bug in your code:**

1. Make sure you pass the 3DES authentication key in the **correct byte order** to your encrypt/decrypt DES-EDE-CBC (Two key triple DES EDE in CBC mode) functions.
   * The Ultralight C authentication key is 16 bytes, where the first 8 bytes (0-7) correspond to Key 1 (K1), and the second 8 bytes (8-15) correspond to Key 2 (K2).
   * The problem is that the required **byte order of the individual keys (K1, K2)** differs based on the context where the key is used (“little-endian” when writing to them to the memory, “BIG-endian” when using them in Node.js `crypto.createDecipheriv`/`crypto.createCipheriv`).
   * For example, the default factory key of MIFARE Ultralight C `Buffer.from('BREAKMEIFYOUCAN!', 'utf-8')` has to be passed as `Buffer.from('IEMKAERB!NACUOYF', 'utf-8')` to the Node.js crypto functions.
   * See my [`MifareUltralightC.swapKeyEndianness()`](https://github.com/pokusew/nfc-pcsc/blob/master/examples/mifare-ultralight-c.js#L299-L329) function, which includes detailed explanations.
   * While developing my implementation, **I also received the `D5-43-02-90-00`[^1] as the response** to the AUTHENTICATE part 2 command (sending `ek(RndA || RndB')`) consistently every time I used the incorrect key. _At first, I was sure my key was correct, so I thought the error must be elsewhere, but then I discovered the tricky byte order thing._
   * Note that if you use a symmetric auth key (resp. an auth key with symmetric K1 and K2 keys) (e.g., all 00 or all FF), the byte order requirement might go unnoticed. That's actually what helped me to uncover the problem, as my implementation worked for all `00` or all `FF` keys, but not for the default `BREAKMEIFYOUCAN!` key.

2. Make sure you use **a correct IV** in your encrypt/decrypt DES-EDE-CBC (Two key triple DES EDE in CBC mode) functions. 
    * The **first** encryption/decryptions uses the **all zero IV**.
       For the **subsequent** encryptions/decryptions, **the IV must be the last ciphertext block**. 
    * It means that during the whole 3DES authentication, your code uses 3 different IVs. See my implementation for reference:
       * [iv1 on line 255](https://github.com/pokusew/nfc-pcsc/blob/master/examples/mifare-ultralight-c.js#L255)
       * [iv2 on line 265](https://github.com/pokusew/nfc-pcsc/blob/master/examples/mifare-ultralight-c.js#L265)
       * [iv3 on line 279](https://github.com/pokusew/nfc-pcsc/blob/master/examples/mifare-ultralight-c.js#L279)
    * During my testing, incorrect IVs resulted in an incorrect RndA being received from the Ultralight C. Surprisingly, the AUTHENTICATE part 2 command succeeded even though I expected it to fail with the `D5-43-02-90-00` error as if an incorrect key had been used. Nevertheless, you have to use the correct IVs in order to be able to tell that the authentication process was really successful (RndA values match).

3. To answer your question:

    > Do I need to call any specific command to wake up or activate the card before starting the authentication process?

    No, you don't. At least, it works for me on macOS with the ACR122U USB NFC reader. **Especially** if you are getting responses to the AUTHENTICATE part 1 command in the expected format `D5-43-00-AF-xx-xx-xx-xx-xx-xx-xx-xx-90-00` (14 bytes, where `xx` is the 8-byte `ek(RndB)`) (you said you are), then it means the authentication process successfully starts, and you should be able to send the AUTHENTICATE part 2 command (and get a successful response) without any problems as well.

    The only other problems I can think of are **(a)** that there might be a too-long delay between the AUTHENTICATE part 1 command and the AUTHENTICATE part 2 command and/or **(b)** that some other process in the OS is communicating with the MIFARE Ultralight C at the same time, thereby interrupting/resetting the authentication process.

   Regarding **(a)**: I tested it, and anything under 1000 ms is fine. Normally, it takes ~60 ms from the point before the AUTHENTICATE part 1 command is sent to the point when the response to the AUTHENTICATE part 2 command is received (including everything, [... cut ...] y every time I used the incorrect 3DES key. Even though the error code might seem misleading, I think it is explainable with the following logic: Incorrect 3DES key -> incorrectly computed `ek(RndA || RndB')` -> Ultralight C compares RndB(s) and finds out they differ -> verification fails, Ultralight C stops the authentication procedure and returns an error, resp., it does not return the normal success response with a CRC, therefore the error is reported as a CRC error (`0x02`) by the NXP PN533.

noproto, 2024-12-24:
@pokusew Awesome work! Thank you.

dongmai, 2024-12-24:
Hi @pokusew, I really appreciate your effort.

I think I might now understand what was wrong with my logic—I was using the incorrect byte order for the key in my encrypt/decrypt logic.

Thank you again!