pokusew/nfc-pcsc #61: Enable Traivs for this repo

amilajack, 2018-12-27:
Travis is currently not enabled for this repo

https://travis-ci.org/pokusew/nfc-pcsc

closed 2018-12-28 by pokusew

pokusew, 2018-12-28:
Hi @amilajack,

I finished [.travis.yml](https://github.com/pokusew/nfc-pcsc/blob/master/.travis.yml) config, enabled Travis CI for the repo and added build status badge to the README.

Next steps:
- [ ] Write tests (start with at least some, e.g. test the static methods of the Reader class)
- [ ] Run tests on Windows and macOS too (consider switching to another CI provider to unify the configs)

Any contribution is welcome! 😉

---

PS Don't forget to star ⭐️my library, if you find it useful. 😃Thanks.

amilajack, 2018-12-28:
> PS Don't forget to star ⭐️my library, if you find it useful. 😃Thanks.

Already did a while ago! 😄

commit 38b4bdb4 (2018-12-28): Add build status badge in README (close #61)