pokusew/nfc-pcsc #65: Cannot process ISO 14443-4 tag because AID was not set

raftheunis87, 2019-01-16:
Hey all,

I'm running an application with the nfc-pcsc module. When I hold my Android device close to the acr-122u, I get the following logging:

`ACS ACR122U PICC Interface  device attached
ACS ACR122U PICC Interface  an error occurred Error: Cannot process ISO 14443-4 tag because AID was not set.
    at ACR122Reader.handle_Iso_14443_4_Tag (/Users/raftheunis/Development/git/DISC/nfc-reader/node_modules/nfc-pcsc/dist/Reader.js:566:26)
    at ACR122Reader.handleTag (/Users/raftheunis/Development/git/DISC/nfc-reader/node_modules/nfc-pcsc/dist/Reader.js:506:21)
    at CardReader.Reader.reader.on (/Users/raftheunis/Development/git/DISC/nfc-reader/node_modules/nfc-pcsc/dist/Reader.js:164:18)
    at <anonymous>
ACS ACR122U PICC Interface  card removed { atr: <Buffer 3b 80 80 01 01>,
  standard: 'TAG_ISO_14443_4',
  type: 'TAG_ISO_14443_4' }`

Any idea how to fix this?

Kind regards!

clementlacroix4, 2019-01-29:
Same issue here. See if you have libnfc-pn53x-examples, libnfc-examples, libnfc-bin. After uninstall issue resolved for me. Maybe a confict between packages

danielzzz, 2019-06-05:
the same here, removing libnfc related stuff didn't help...

pokusew, 2019-06-06:
Hi @danielzzz,

what are you trying to achieve? What card/tag do you use? Do you want to communicate with Android device via [Android HCE](https://developer.android.com/guide/topics/connectivity/nfc/hce.html)? 

`Cannot process ISO 14443-4 tag because AID was not set.` error is thrown when the following conditions are met:
1. `reader.autoProcessing` is enabled (true) (that's by default)
2. `reader.aid` is not set
3. ISO 14443-4 tag is detected (e.g. MIFARE DESFire, Android NFC enabled device, ...)

You have two options:
1. Disable autoProcessing by setting `reader.autoProcessing = false` (this affects all tag types).
  Then you can process the tags yourself (i.e. send custom commands using `reader.transmit`).
   **See [Alternative usage](https://github.com/pokusew/nfc-pcsc#alternative-usage) section** in the project's README for more info on this approach.
  BTW: this way [MIFARE DESFire example](https://github.com/pokusew/nfc-pcsc/blob/master/examples/mifare-desfire.js) is implemented
2. In case you OK with and want to do autoProcessing the ISO 14443-4 tags by sending SELECT FILE command with the AID, then you have to set `reader.aid` as described in [basic example on the line 17](https://github.com/pokusew/nfc-pcsc/blob/master/examples/basic.js#L17).

**In case you are trying to use Android HCE**, the second option could work for you. You set the `reader.aid` to some custom unique HEX string which will identify your app. Then you set the same value in `aid-filter` tag in the `host-apdu-service` manifest as described in [Service manifest declaration and AID registration](https://developer.android.com/guide/topics/connectivity/nfc/hce.html#ManifestDeclaration) in Android HCE guide. Then, when a NFC enabled Android device with you HCE service approaches the NFC reader and is detected, nfc-pcsc sends SELECT FILE command with your AID. This command tells Android OS that all further commands (sent using `reader.transmit`) should be send to your app's [HCE service](https://developer.android.com/guide/topics/connectivity/nfc/hce.html#ServiceImplementation). This way you can implement bi-directional communication with Android NFC device. Refer to the [Android HCE docs](https://developer.android.com/guide/topics/connectivity/nfc/hce.html#HCE) for more info.

Hope it helps. 🙂 

---

PS Don't forget to star ⭐️my library, if you find it useful. 😃Thanks.

---

cc @raftheunis87 It may be helpful to you, as well. 🙂  Sorry for not responding earlier.

danielzzz, 2019-06-07:
hi @pokusew - thanks for this extended answer. I really appreciate it!

I had this reader laying around and I wanted to check if it is capable of reading nfc tags from my phone.

thanks for the pointers to the docs I will check it out if it helps.

cheers!
dan

danielzzz, 2021-01-04:
hi, ignore my comment, I had a wrong fake reader which didn't work properly. using an acr122 worked fine.
I added a pull request for react-native-hce if you find it might be interesting to add.

best, dan

M4Tdev, 2024-11-13:
Hey @pokusew, I want to read the uid of the card which happens to be the ISO 14443-4 card, it is an SEOS IP card, do you happen to know what would be the AID for this card or how could I find it basically, also I am using the Omnikey 5427 card reader