pokusew/nfc-pcsc #67: Unique ID

ollietb, 2019-01-28:
Hi there
I'm creating an Electron app with 4 NFC readers attached. I am wondering whether there is a way of getting a unique ID per reader, so that I can tell which read has come from which device, even if I unplug and re-plug them back into a different USB port? At the moment I can see that the only unique identifier is the "Interface 1", "Interface 2" on the reader object, but that will change if I change the USB ports.

celevra, 2019-04-18:
have you ever found a solution to that problem?
where did you get the interface numbers from?

ollietb, 2019-04-18:
No we had to go with a different solution in the end

andreykats, 2019-12-17:
This is not really a good solution but nfc-pcsc does assign unique device name when attaching additional devices.
[image]

The names do get reassigned when reattaching the usb devices the devices so we need to find a better solution.

vaaski, 2024-09-14:
Just had a similar issue, this might help.
https://github.com/pokusew/nfc-pcsc/issues/130#issuecomment-2351019089