pokusew/nfc-pcsc #69: Detect acr122l USB version with LCD

Ionut-Milas, 2019-02-04:
Hi @pokusew !
First of all I want to thank you for this great library.
I am using in another project and it works great.
I have a new use case where I need to connect to and ACR122L - USB version ( https://www.acs.com.hk/en/products/144/acr122l-visualvantage-serial-nfc-reader-with-lcd/ )

However the lib does not detect the reader.
I am able to detect the reader with this library: `usb-detection`.
Here is the output:
```
{ locationId: 336592896,
    vendorId: 1839,
    productId: 8710,
    deviceName: 'ACR1222L 3S CL Reader',
    manufacturer: 'ACS',
    serialNumber: '',
    deviceAddress: 27 }
```
I know that supporting this device in your library is a far fetch, but perhaps you can offer some guidance on how to proceed.

Thank you!

craftbyte, 2019-02-10:
Hey there,
Do you see the reader as a PCSC device?
Thanks

Franx0, 2019-02-13:
@Ionut-Milas have you installed all the needed drivers (https://www.acs.com.hk/en/products/144/acr122l-visualvantage-serial-nfc-reader-with-lcd/#downloads). I guess so cause another project of yours is working fine but I'm working with that device and it's working fine, it's weird.

Ionut-Milas, 2019-02-13:
Hi guys,
It detected it after I installed the drivers.
@Franx0 were you able to control the LED's and LCD scroll ?
i am able to read/write a card and display a message on the LCD.