pokusew/nfc-pcsc #74: Unable to install nfc-pcsc on Raspbian anymore

unknowndomain, 2019-03-26:
I tried installing nfc-pcsc using NPM on Node 8 and Node 10 with the same result, this is in  blank folder with a blank package.json trying to install using this command for additional information `$ npm install -d nfc-pcsc`.

````
npm info it worked if it ends with ok
npm info using npm@6.4.1
npm info using node@v10.15.3
npm http fetch GET 304 https://registry.npmjs.org/nfc-pcsc 333ms (from cache)
npm timing stage:loadCurrentTree Completed in 1151ms
npm timing stage:loadIdealTree:cloneCurrentTree Completed in 2ms
npm timing stage:loadIdealTree:loadShrinkwrap Completed in 16ms
npm http fetch GET 304 https://registry.npmjs.org/@pokusew%2fpcsclite 840ms (from cache)
npm http fetch GET 304 https://registry.npmjs.org/bindings 68ms (from cache)
npm http fetch GET 304 https://registry.npmjs.org/nan 117ms (from cache)
npm http fetch GET 304 https://registry.npmjs.org/file-uri-to-path 52ms (from cache)
npm timing stage:loadIdealTree:loadAllDepsIntoIdealTree Completed in 1105ms
npm timing stage:loadIdealTree Completed in 1143ms
npm timing stage:generateActionsToTake Completed in 23ms
npm timing audit compress Completed in 13ms
npm info audit Submitting payload of 662bytes
npm timing audit submit Completed in 481ms
npm http fetch POST 200 https://registry.npmjs.org/-/npm/v1/security/audits/quick 480ms
npm timing audit body Completed in 3ms
npm timing action:extract Completed in 459ms
npm timing action:finalize Completed in 28ms
npm timing action:refresh-package-json Completed in 82ms
npm info lifecycle file-uri-to-path@1.0.0~preinstall: file-uri-to-path@1.0.0
npm info lifecycle bindings@1.5.0~preinstall: bindings@1.5.0
npm info lifecycle nan@2.13.2~preinstall: nan@2.13.2
npm info lifecycle @pokusew/pcsclite@0.5.2~preinstall: @pokusew/pcsclite@0.5.2
npm info lifecycle nfc-pcsc@0.7.1~preinstall: nfc-pcsc@0.7.1
npm timing action:preinstall Completed in 15ms
npm info linkStuff file-uri-to-path@1.0.0
npm info linkStuff bindings@1.5.0
npm info linkStuff nan@2.13.2
npm info linkStuff @pokusew/pcsclite@0.5.2
npm info linkStuff nfc-pcsc@0.7.1
npm timing action:build Completed in 15ms
npm info lifecycle file-uri-to-path@1.0.0~install: file-uri-to-path@1.0.0
npm info lifecycle bindings@1.5.0~install: bindings@1.5.0
npm info lifecycle nan@2.13.2~install: nan@2.13.2
npm info lifecycle @pokusew/pcsclite@0.5.2~install: @pokusew/pcsclite@0.5.2

> @pokusew/pcsclite@0.5.2 install /home/pi/test/node_modules/@pokusew/pcsclite
> node-gyp rebuild

gyp info it worked if it ends with ok
gyp info using node-gyp@3.8.0
gyp info using node@10.15.3 | linux | arm
gyp info spawn /usr/bin/python2
gyp info spawn args [ '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/home/pi/test/node_modules/@pokusew/pcsclite/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/pi/.node-gyp/10.15.3/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/home/pi/.node-gyp/10.15.3',
gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/home/pi/.node-gyp/10.15.3/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/home/pi/test/node_modules/@pokusew/pcsclite',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/home/pi/test/node_modules/@pokusew/pcsclite/build'
  CXX(target) Release/obj.target/pcsclite/src/addon.o
In file included from ../src/addon.cpp:1:0:
../src/pcsclite.h:9:22: fatal error: winscard.h: No such file or directory
 #include <winscard.h>
                      ^
compilation terminated.
pcsclite.target.mk:117: recipe for target 'Release/obj.target/pcsclite/src/addon.o' failed
make: *** [Release/obj.target/pcsclite/src/addon.o] Error 1
make: Leaving directory '/home/pi/test/node_modules/@pokusew/pcsclite/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.14.79-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/test/node_modules/@pokusew/pcsclite
gyp ERR! node -v v10.15.3
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm info lifecycle @pokusew/pcsclite@0.5.2~install: Failed to exec install script
npm timing action:install Completed in 2709ms
npm timing stage:rollbackFailedOptional Completed in 113ms
npm timing stage:runTopLevelLifecycles Completed in 5957ms
npm WARN test@1.0.0 No description
npm WARN test@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @pokusew/pcsclite@0.5.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @pokusew/pcsclite@0.5.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm timing npm Completed in 7269ms

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-03-26T14_28_50_024Z-debug.log
````

unknowndomain, 2019-03-28:
Turns out I overlooked the need to:
````
sudo apt-get install libpcsclite1 libpcsclite-dev
sudo apt-get install pcscd
````

zacharycohn, 2020-03-11:
I'm having a different issue installing on Raspbian. I go through the installation steps, do "npm run example" and I get the following output: 

> nfc-pcsc@0.8.0 example /home/pi/SoundCube/nodeNFC/nfc-pcsc
> node -r @babel/register examples/read-write.js

9:39:02 PM – ACS ACR 38U-CCID 00 00 info: device attached
9:39:03 PM – ACS ACR 38U-CCID 00 00 error: an error occurred Error: Not found response. Tag not compatible with AID F222222222.
    at Reader.handle_Iso_14443_4_Tag (/home/pi/SoundCube/nodeNFC/nfc-pcsc/src/Reader.js:767:17)

Not sure where to go from here...