pokusew/nfc-pcsc #78: example-mifare-desfire only one appId

celevra, 2019-04-11:
Hi,

i tried the example with these settings
```
const desfire = {
	key: 'xxxx',
	appId: [0x17],
	keyId: [7],
};
```

but i only get an error and i can't really tell why, could you guide me in the right direction?

```
4:08:33 PM – ACS ACR122U 00 00 info: device attached
4:08:34 PM – ACS ACR122U 00 00 info: card detected { atr: <Buffer 3b 81 80 01 80 80>,
  standard: 'TAG_ISO_14443_4',
  type: 'TAG_ISO_14443_4' }
[step 1 - select app] sending <Buffer 90 6a 00 00 01 17 00>
[step 1 - select app] received data <Buffer 91 7e>
4:08:34 PM – ACS ACR122U 00 00 error: error occurred during processing steps Error: error in step 1
    at selectApplication (/root/node/nfc-pcsc/examples/mifare-desfire.js:84:11)

```

celevra, 2019-04-13:
problem solved, it has to be 3 appId's so the example is totally correct, sorry for the inconvenience

closed 2019-04-13 by celevra

anupamjuit, 2021-08-20:
Hi! 
I am facing this problem, could you help me:

8:47:14 pm – ACS ACR1252 Dual Reader PICC 0 info: card detected {
  atr: <Buffer 3b 81 80 01 80 80>,
  standard: 'TAG_ISO_14443_4',
  type: 'TAG_ISO_14443_4'
}
[step 1 - select app] sending <Buffer 90 5a 00 00 03 00 00 00 00>
[step 1 - select app] received data <Buffer 91 00>
[step 2 - authenticate] sending <Buffer 90 0a 00 00 01 00 00>
[step 2 - authenticate] received data <Buffer 8c 91 b3 27 ba d2 73 7c 91 af>
[step 2 - set up RndA] sending <Buffer 90 af 00 00 10 ed ed ba 27 8b 27 ab 77 46 a9 48 01 c8 1e e6 e3 00>
[step 2 - set up RndA] received data <Buffer 05 c7 89 14 c6 89 fd 02 91 00>
[step 3 - read] sending <Buffer 90 bd 00 00 07 02 00 00 00 0e 00 00 00>
[step 3 - read] received data <Buffer 91 9d>
8:47:14 pm – ACS ACR1252 Dual Reader PICC 0 error: error occurred during processing steps Error: error in step 3 - read
    at readData (C:\Users\Troveto\Downloads\nfc-pcsc-master (1)\nfc-pcsc-master\examples\/mifare-desfire.js:160:11)
    at ACR122Reader.<anonymous> (C:\Users\Troveto\Downloads\nfc-pcsc-master (1)\nfc-pcsc-master\examples\/mifare-desfire.js:178:4)