pokusew/nfc-pcsc #82: Angular 7 installation

Canos, 2019-05-03:
i'm trying to use in a regular angular 7 app.

following installation notes: I've tried 
`yarn add nfc-pcsc` 

After using this import 
import { NFC } from 'nfc-pcsc';

I get a 
```
ERROR in ./node_modules/bindings/bindings.js
Module not found: Error: Can't resolve 'fs' in '****/node_modules/bindings'
ERROR in ./node_modules/bindings/bindings.js
Module not found: Error: Can't resolve 'path' in '****/node_modules/bindings'
ERROR in ./node_modules/file-uri-to-path/index.js
Module not found: Error: Can't resolve 'path' in '****/node_modules/file-uri-to-path'
```

any official solution or workaround? 
Thanks

cichybiala, 2020-07-14:
@Canos 
Did you found any solution?

Canos, 2020-07-14:
No