pokusew/nfc-pcsc #84: new NFC() hangs on windows 10

fatmatto, 2019-05-28:
Hello,

I tried the example on the README file on a windows 10 tablet and it **hangs** on the line:

```javascript
const nfc = new NFC(); 
```

Inspecting the code in the node_modules/nfc-pcsc/dist directory I found out that it blocks at the line following line in dist/NFC.js 

```javascript
this.pcsc = (0, _pcsclite.default)()
```

Which in turns blocks at the line

```jaascript
const p = new PCSCLite()
````
in @pokusew/pcsclite/lib/pcsclite.js

Is there anything that is required to make the library work on Windows 10?

The system configuration is
```
node v10.15.3
npm 6.4.1
Windows 10 Enterprise 
```

pokusew, 2019-05-28:
Hi @fatmatto, it is known issue on Windows that can occur in some situations. See https://github.com/pokusew/node-pcsclite/issues/21.

> The example was run on a Windows 10 (Pro) machine that had never had a smart card device plugged in to it. The example seems to run fine on Win 10 machines that have seen a smart card device.

Smart Card Service on Windows is not running by default and is only started when a NFC reader is connected (I tried to change this behavior, but does not seem to be possible). This had been causing pcsclite (nfc-pcsc) to crash. I (my friend on Windows, I use macOS) did some fixes (https://github.com/pokusew/node-pcsclite/commit/9f871e90966d852e440ff62025ca4c70f72c8667) that seemed to fix the issue. Apparently, now in some configurations (e.g. on a machine that had never had a smart card device plugged in to it), the issue you describing occurs.

There is already a PR https://github.com/pokusew/node-pcsclite/pull/22 that is addressing this issue. Unfortunately, I haven't got enough time yet to examine it and eventually merge the fix.

Hope it helps. 🙂

---

As soon as I get some time (hopefully in a month or so), I try to look into this and merge the PR.

fatmatto, 2019-05-29:
Hi @pokusew Thank you for the reply, since i already used the NFC on that device (with a Windows native app) I assumed it counted as a smart card device previously plugged in :) 

While you handle the PR i'll take a look at it and try with a fork of the repo.

Thank you for the repo and for your time :)

cloudhx, 2020-10-28:
Hello,

I have the same issue with windows 10, I tried to manually start the `Smart Card` service, but it didn't help.

Is this something that can be fixed?

Thank you very much!

kumarmboxuk, 2021-06-02:
Can you explain how did you fix this issue please? :)

Fossil01, 2021-06-23:
I am running into the same issue and it's unclear to me how to fix this.

cloudhx, 2021-07-09:
@Fossil01 @kumarmboxuk I'm using ACR122 reader and as I remembered you need to install the driver for it on Windows.

esskar, 2024-04-26:
The hanging also happens, when trying to run `new NFC()` while conneting via remote desktop

adlanarifzr, 2024-04-26:
> The hanging also happens, when trying to run `new NFC()` while conneting via remote desktop

You need to make sure the windows device connected to the reader.