pokusew/nfc-pcsc #86: Local Example Install Errors

nchiles, 2019-06-12:
I'm unable to install the example files locally. I've tried the solutions in [this thread](https://github.com/nodejs/node-gyp/issues/809) with no luck. Any ideas to fix the errors?

> @pokusew/pcsclite@0.5.2 install /Users/natechiles/nfc-pcsc/node_modules/@pokusew/pcsclite
> node-gyp rebuild

  CXX(target) Release/obj.target/pcsclite/src/addon.o
In file included from ../src/addon.cpp:1:
../src/pcsclite.h:31:30: error: no template named 'Handle' in namespace 'v8'
        static void init(v8::Handle<v8::Object> target);
                         ~~~~^
In file included from ../src/addon.cpp:2:
../src/cardreader.h:90:30: error: no template named 'Handle' in namespace 'v8'
        static void init(v8::Handle<v8::Object> target);
                         ~~~~^
../src/cardreader.h:123:20: error: no template named 'Handle' in namespace 'v8'
        static v8::Handle<v8::Value> CreateBufferInstance(char* data, un...
               ~~~~^
../src/addon.cpp:4:6: error: variable has incomplete type 'void'
void init_all(v8::Handle<v8::Object> target) {
     ^
../src/addon.cpp:4:36: error: expected '(' for function-style cast or type
      construction
void init_all(v8::Handle<v8::Object> target) {
                         ~~~~~~~~~~^
../src/addon.cpp:4:19: error: no member named 'Handle' in namespace 'v8'
void init_all(v8::Handle<v8::Object> target) {
              ~~~~^
../src/addon.cpp:4:38: error: use of undeclared identifier 'target'
void init_all(v8::Handle<v8::Object> target) {
                                     ^
../src/addon.cpp:4:45: error: expected ';' after top level declarator
void init_all(v8::Handle<v8::Object> target) {
                                            ^
                                            ;
8 errors generated.
make: *** [Release/obj.target/pcsclite/src/addon.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:200:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Darwin 18.6.0
gyp ERR! command "/usr/local/Cellar/node/12.4.0/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/natechiles/nfc-pcsc/node_modules/@pokusew/pcsclite
gyp ERR! node -v v12.4.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @pokusew/pcsclite@0.5.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @pokusew/pcsclite@0.5.2 install script.

nchiles, 2019-06-12:
same issue here `https://github.com/pokusew/node-pcsclite/issues/24`

closed 2020-01-25 by pokusew

pokusew, 2020-01-26:
@nchiles Fixed and released in version [0.8.0](https://github.com/pokusew/nfc-pcsc/releases/tag/v0.8.0). 🎉