pokusew/nfc-pcsc #89: Issue with card reader previously attached

elia9090, 2019-07-12:
Hi,
first of all, thanks for your library.
I'm using your library on an Electron app with an ACR122U card reader and Mirafare Classic tags.
Everything works perfectly **but if the app starts and the reader is already attached to the PC it does not register it**.

I tried to use the usb-detection library and the device is correctly shown. 

Can you help me?

mathiscode, 2019-11-25:
+1; I'm not using Electron, but I have a similar issue. I have to physically disconnect/reconnect the reader before the nfc.on('reader') listener gets fired.

mlongo4290, 2020-02-05:
+1; Same issue here. I'm on an electron app.

mlongo4290, 2020-02-06:
I've found out!!
Instantating nfc right after the require lead to the error we reported. So instead of doing this
```
const { NFC } = require('nfc-pcsc');
const nfc = new NFC(); //THIS LEAD TO OUR ERROR
//other code
app.on('ready', function() {
    //your code
    nfc.on('reader', reader => {//handle reader});
})
```

We must do this
```
const { NFC } = require('nfc-pcsc');
//other code
app.on('ready', function() {
    //your code
    const nfc = new NFC(); //THIS HANDLES ALREADY CONNECTED READER IN THE RIGHT WAY
    nfc.on('reader', reader => {//handle reader});
})
```