pokusew/nfc-pcsc #90: No card events when placing/removing card on Windows 10 with ACS ACR1281U Electron

markhorsman, 2019-07-14:
Works perfectly in Linux but when I finally got it working on Windows it only shows when a reader is connected/disconnected, no card events.

Node 10.16.0
Yarn 1.17.3
Electron 3.0
Windows 10 x64
Reader: ACS ACR1281U

3dmedium, 2019-10-17:
I encounter the same error after windows updates (Enterprise win so late updates, cant tell which exactly)

There is a software calles Spring Card Diagnostic from https://www.springcard.com/en/download/find/file/pcsc-sdk
Which for me indicates the Reader working or not. THis software also had no luck to controll the Reader (Card events worked but no commands)

For me The PC/SC Diag worked after downgrading from Driver Version 4.2.8.0 to 4.0.6.0
But still nfc-pcsc did not.

Reader for me is ACR1252

3dmedium, 2019-10-17:
Recompiling with pcsclite which has the native parts for nfc-pcsc with the older driver 4.0.6.0 fixed it for me.

Think this is only a workaround. Driver 4.2.8.0 did not work after recompiling but is working on a machine with Windows 1803