pokusew/nfc-pcsc #91: erro install electron app and build native

fzanfolim, 2019-10-23:
when installing lib pokusew / nfc-pcsc and running the electron project, I got the error: (node_modules/@pokusew/pcsclite/build/Release/pcsclite.node '
was compiled against a different Node.js version using
NODE_MODULE_VERSION 64. This version of Node.js requires
NODE_MODULE_VERSION 73. Please try re-compiling or re-installing.
the module (for instance, using `npm rebuild` or` npm install`).
    at process.func (electron / js2c / asar.js: 138)
    at process.func [as dlopen] (electron / js2c / asar.js: 138)
    at Object.Module._extensions..node (internal / modules / cjs / loader.js: 828)
    at Object.func (electron / js2c / asar.js: 138)
    at Object.func [as .node] (electron / js2c / asar.js: 138)
    at Module.load (internal / modules / cjs / loader.js: 645)
    at Function.Module._load (internal / modules / cjs / loader.js: 560)
    at Module.require (internal / modules / cjs / loader.js: 685)
    at require (internal / modules / cjs / helpers.js: 16)
    at bindings (/Users/felipezanfoli…ngs/bindings.js:112)), so I run the electron build: ./node_modules/.bin/electron-rebuild and there is an error in the command as per attachment.

[image]

betrri, 2019-11-08:
Those Handle objects were removed/renamed with newer versions of v8 javascript engine. I recommend  using this repo instead: https://github.com/ap-mitch/nfc-pcsc#readme

closed 2020-01-25 by pokusew

pokusew, 2020-01-26:
@fzanfolim @betrri Fixed and released in version [0.8.0](https://github.com/pokusew/nfc-pcsc/releases/tag/v0.8.0). 🎉