pokusew/nfc-pcsc #92: Can't install on Windows 10 -- 'Handle': is not a member of 'v8'

bradisbell, 2019-10-26:
Can't seem to get this installed on Windows.  Any pointers would be helpful.

```
yarn add nfc-pcsc
yarn add v1.13.0
[1/4] Resolving packages...
[2/4] Fetching packages...
info fsevents@1.2.9: The platform "win32" is incompatible with this module.
info "fsevents@1.2.9" is an optional dependency and failed compatibility check. Excluding it from installation.
[3/4] Linking dependencies...
[4/4] Building fresh packages...
error node_modules\@pokusew\pcsclite: Command failed.
Exit code: 1
Command: node-gyp rebuild
Arguments:
Directory: node_modules\@pokusew\pcsclite
Output:
node_modules\@pokusew\pcsclite>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "" rebuild )
gyp info it worked if it ends with ok
gyp info using node-gyp@5.0.3
gyp info using node@12.11.1 | win32 | x64
gyp info find Python using Python version 2.7.16 found at "C:\Python27\python.exe"
gyp info find VS using VS2017 (15.9.28307.858) found at:
gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools"
gyp info find VS run with --verbose for detailed information
gyp info spawn C:\Python27\python.exe
gyp info spawn args [
gyp info spawn args   'C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-I',
gyp info spawn args   'node_modules\\@pokusew\\pcsclite\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'AppData\\Local\\node-gyp\\Cache\\12.11.1\\include\\node\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=AppData\\Local\\node-gyp\\Cache\\12.11.1',
gyp info spawn args   '-Dnode_gyp_dir=C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp',
gyp info spawn args   '-Dnode_lib_file=AppData\\Local\\node-gyp\\Cache\\12.11.1\\<(target_arch)\\node.lib',
gyp info spawn args   '-Dmodule_root_dir=node_modules\\@pokusew\\pcsclite',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'node_modules\\@pokusew\\pcsclite\\build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe
gyp info spawn args [
gyp info spawn args   'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64'
gyp info spawn args ]
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  addon.cpp
  pcsclite.cpp
  cardreader.cpp
  win_delay_load_hook.cc
node_modules\@pokusew\pcsclite\src\pcsclite.h(31): error C2039: 'Handle': is not a member of 'v8' (compiling source file ..\src\pcsclite.cpp) [node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  appdata\local\node-gyp\cache\12.11.1\include\node\v8-platform.h(16): note: see declaration of 'v8' (compiling source file ..\src\pcsclite.cpp)
node_modules\@pokusew\pcsclite\src\pcsclite.h(31): error C2061: syntax error: identifier 'Handle' (compiling source file ..\src\pcsclite.cpp) [node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
node_modules\@pokusew\pcsclite\src\cardreader.h(90): error C2039: 'Handle': is not a member of 'v8' (compiling source file ..\src\cardreader.cpp) [node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  appdata\local\node-gyp\cache\12.11.1\include\node\v8-platform.h(16): note: see declaration of 'v8' (compiling source file ..\src\cardreader.cpp)
node_modules\@pokusew\pcsclite\src\cardreader.h(90): error C2061: syntax error: identifier 'Handle' (compiling source file ..\src\cardreader.cpp) [node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
node_modules\@pokusew\pcsclite\src\cardreader.h(123): error C2039: 'Handle': is not a member of 'v8' (compiling source file ..\src\cardreader.cpp) [node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  appdata\local\node-gyp\cache\12.11.1\include\node\v8-platform.h(16): note: see declaration of 'v8' (compiling source file ..\src\cardreader.cpp)
[... 75 lines cut ...]
node_modules\@pokusew\pcsclite\src\cardreader.cpp(534): warning C4996: 'v8::Object::Set': was declared deprecated [node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  appdata\local\node-gyp\cache\12.11.1\include\node\v8.h(3402): note: see declaration of 'v8::Object::Set'
node_modules\@pokusew\pcsclite\src\cardreader.cpp(540): warning C4996: 'Nan::Callback::Call': was declared deprecated [node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  documents\beatlink-token-tool\node_modules\nan\nan.h(1740): note: see declaration of 'Nan::Callback::Call'
node_modules\@pokusew\pcsclite\src\cardreader.cpp(593): warning C4996: 'Nan::Callback::Call': was declared deprecated [node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  documents\beatlink-token-tool\node_modules\nan\nan.h(1740): note: see declaration of 'Nan::Callback::Call'
node_modules\@pokusew\pcsclite\src\cardreader.cpp(601): warning C4996: 'Nan::Callback::Call': was declared deprecated [node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  documents\beatlink-token-tool\node_modules\nan\nan.h(1740): note: see declaration of 'Nan::Callback::Call'
node_modules\@pokusew\pcsclite\src\cardreader.cpp(657): warning C4996: 'Nan::Callback::Call': was declared deprecated [node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  documents\beatlink-token-tool\node_modules\nan\nan.h(1740): note: see declaration of 'Nan::Callback::Call'
node_modules\@pokusew\pcsclite\src\cardreader.cpp(665): warning C4996: 'Nan::Callback::Call': was declared deprecated [node_modules\@pokusew\pcsclite\build\pcsclite.vcxproj]
  documents\beatlink-token-tool\node_modules\nan\nan.h(1740): note: see declaration of 'Nan::Callback::Call'
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:196:23)
gyp ERR! stack     at ChildProcess.emit (events.js:210:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Windows_NT 10.0.18362
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd node_modules\@pokusew\pcsclite
gyp ERR! node -v v12.11.1
gyp ERR! node-gyp -v v5.0.3
gyp ERR! not ok
info Visit https://yarnpkg.com/en/docs/cli/add for documentation about this command.
```

AlexQuidditch, 2019-11-14:
@bradisbell i have same issue

dbauducco, 2020-01-20:
Did you ever find a fix?

crazyquark, 2020-01-23:
not a solution, but if you downgrade your node to say v10, it will compile,

closed 2020-01-25 by pokusew

pokusew, 2020-01-26:
@bradisbell @AlexQuidditch @dbauducco @crazyquark  Fixed and released in version [0.8.0](https://github.com/pokusew/nfc-pcsc/releases/tag/v0.8.0). 🎉

crazyquark, 2020-04-01:
Indeed, can confirm that it now works.

riyupatel, 2020-09-02:
@pokusew  i start "Smart Card" and "Smart Card Device Enumeration Service" in services on Windows 10 , but even so i face this error
C:\node_pro\node_modules\pcsclite\lib\pcsclite.js:39
var p = new PCSCLite(); ^ Error: SCardEstablishContext error: The Smart Card Resource Manager is not running. (0x8010001d)

card reader : verifone vx 520 (connected via USB Serial Device(COM3))
can you help me ? thank you