pokusew/nfc-pcsc #95: Typescript Definition

nicklaros, 2019-12-04:
Hi, I'm trying to use this library in typescript project but got error `Could not find a declaration file for module 'nfc-pcsc'`

Could you please add typescript definition to this awesome library. thanks.

vaaski, 2021-06-20:
I've just started working on this, so far I've covered enough for my very small use case of just getting the UID from a generic card.

I don't really know all the technicalities of all the card types and such neither do I know the inner workings of `@pokusew/pcsclite`, but I'd be happy to help implement types for this repo.

I've added the types in a way in which no source code is modified, and the definitions in [Reader.d.ts](https://github.com/vaaski/nfc-pcsc/blob/3365c5777571c1da0845cfcaa0252b2e4e7de936/src/Reader.d.ts) should be easy to work with and extend.

My fork is [here](https://github.com/vaaski/nfc-pcsc).

To test the types you can add `// @ts-check` to the top of an example file to enable strict type-checking (at least in vscode).

Not sure if you got resources to work on this right now @pokusew, but I'd be willing to help :)

bastianandre, 2022-03-06:
Thanks vaaski for sharing your types! 
It would also be great to see your implementation in a module declaration, so it can be retrofittet independant of the repo.
I was not able to get it right as I'm not experienced in TS. My approach:

```
declare module 'nfc-pcsc' {
  import type { TypedEmitter } from './TypedEmitter'
  interface NFCEmitter {
    reader: (reader: Reader) => void;
    error: (error: Error) => void;
  }
  export class NFC extends TypedEmitter<NFCEmitter> {}
}
```

Can someone assist?

davidgs, 2023-04-05:
I wonder if this will work: 

```js
declare module 'nfc-pcsc' {
  import { EventEmitter } from 'events';

  interface NfcPcsc extends EventEmitter {
    start(): Promise<void>;
    stop(): Promise<void>;
    listDevices(): Promise<string[]>;
    open(device: string): Promise<void>;
    close(): Promise<void>;
    transmit(data: Buffer): Promise<Buffer>;
  }

  function nfc(): NfcPcsc;

  export = nfc;
}
```

mizushino, 2023-04-22:
I didn't write it all down, but it looks like this.

nfc-pcsc.d.ts
```ts
declare module 'nfc-pcsc' {
  export type ListenerSignature<L> = {
    [E in keyof L]: (...args: any[]) => any;
  };

  export type DefaultListener = {
    [k: string]: (...args: any[]) => any;
  };

  export class TypedEmitter<L extends ListenerSignature<L> = DefaultListener> {
    static defaultMaxListeners: number;
    addListener<U extends keyof L>(event: U, listener: L[U]): this;
    prependListener<U extends keyof L>(event: U, listener: L[U]): this;
    prependOnceListener<U extends keyof L>(event: U, listener: L[U]): this;
    removeListener<U extends keyof L>(event: U, listener: L[U]): this;
    removeAllListeners(event?: keyof L): this;
    once<U extends keyof L>(event: U, listener: L[U]): this;
    on<U extends keyof L>(event: U, listener: L[U]): this;
    off<U extends keyof L>(event: U, listener: L[U]): this;
    emit<U extends keyof L>(event: U, ...args: Parameters<L[U]>): boolean;
    eventNames<U extends keyof L>(): U[];
    listenerCount(type: keyof L): number;
    listeners<U extends keyof L>(type: U): L[U][];
    rawListeners<U extends keyof L>(type: U): L[U][];
    getMaxListeners(): number;
    setMaxListeners(n: number): this;
  }

  type Type = 'TAG_ISO_14443_3' | 'TAG_ISO_14443_4';

  const KEY_TYPE_A = 0x60;
  const KEY_TYPE_B = 0x61;

  interface Card {
    type: Type;
    standard: Type;
    uid?: string;
    data?: Buffer;
  }

  interface ReaderEmitter {
    card: (x: Card) => void;
    'card.off': (x: Card) => void;
    error: (x: Error) => void;
    end: () => void;
  }

  export class Reader extends TypedEmitter<ReaderEmitter> {
    get name(): string;

    authenticate(blockNumber: number, keyType: number, key: string, obsolete?: boolean): Promise<boolean>;

    read(
      blockNumber: number,
      length: number,
      blockSize?: number,
      packetSize?: number,
      readClass?: number
    ): Promise<Buffer>;
  }

  interface NFCEmitter {
    reader: (reader: Reader) => void;
    error: (error: Error) => void;
  }

  export class NFC extends TypedEmitter<NFCEmitter> {}
}
```