pokusew/nfc-pcsc #97: Error: Not found response. Tag not compatible with AID F222222222

zacharycohn, 2020-03-11:
On Raspbian, I installed pcsc smoothly but when I run "npm start example" I get this: 

> nfc-pcsc@0.8.0 example /home/pi/SoundCube/nodeNFC/nfc-pcsc
> node -r @babel/register examples/read-write.js

9:39:02 PM – ACS ACR 38U-CCID 00 00 info: device attached
9:39:03 PM – ACS ACR 38U-CCID 00 00 error: an error occurred Error: Not found response. Tag not compatible with AID F222222222.
    at Reader.handle_Iso_14443_4_Tag (/home/pi/SoundCube/nodeNFC/nfc-pcsc/src/Reader.js:767:17)

Not sure what that error means. I'm using a Touchatag / acr122.

zacharycohn, 2020-03-11:
I can get a log if I pass in the logger, which outputs: 

> nfc-pcsc@0.8.0 example /home/pi/SoundCube/nodeNFC/nfc-pcsc
> node -r @babel/register examples/read-write.js

10:07:29 PM – debug: new reader detected 'ACS ACR 38U-CCID 00 00'
10:07:29 PM – ACS ACR 38U-CCID 00 00 info: device attached
10:07:29 PM – debug: status {
  state: 34,
  atr: <Buffer 3b be 96 00 00 41 03 00 00 00 00 00 00 00 00 00 02 90 00>
}
10:07:29 PM – debug: changes 34
10:07:29 PM – debug: card inserted <Buffer 3b be 96 00 00 41 03 00 00 00 00 00 00 00 00 00 02 90 00>
10:07:29 PM – debug: trying to connect [ 'CONNECT_MODE_CARD', 2 ]
10:07:29 PM – debug: connected { type: 2, protocol: 1 }
10:07:29 PM – debug: handling tag {
  atr: <Buffer 3b be 96 00 00 41 03 00 00 00 00 00 00 00 00 00 02 90 00>,
  standard: 'TAG_ISO_14443_4',
  type: 'TAG_ISO_14443_4'
}
10:07:29 PM – debug: processing ISO 14443-4 tag {
  atr: <Buffer 3b be 96 00 00 41 03 00 00 00 00 00 00 00 00 00 02 90 00>,
  standard: 'TAG_ISO_14443_4',
  type: 'TAG_ISO_14443_4'
}
10:07:29 PM – debug: transmitting [ <Buffer 00 a4 04 00 05 f2 22 22 22 22 00>, 40 ]
10:07:29 PM – debug: transmit response received [ <Buffer 6a 82>, 2 ]
10:07:29 PM – ACS ACR 38U-CCID 00 00 error: an 1 error occurred Error: Not found response. Tag not compatible with AID F222222222.
    at Reader.handle_Iso_14443_4_Tag (/home/pi/SoundCube/nodeNFC/nfc-pcsc/src/Reader.js:767:17)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)

Sennar4, 2020-04-08:
I have the same problem and no idea what to do. Did you fixed it meanwhile?

zacharycohn, 2020-04-08:
> I have the same problem and no idea what to do. Did you fixed it meanwhile?

Unfortunately no. I never got an answer to this and could never figure it out.

Sennar4, 2020-04-08:
I don't know if its helpful for you but I found this: https://www.npmjs.com/package/simple-pcsc1
Seems like a shorter version but if you install it and uses "npm run test" it works.

Xample, 2020-06-02:
It depends on the card, I've 2 cards, one is a credit card and the second one a white rfid card. 

The credit card gives the error:

```
ACS ACR122U PICC Interface ERROR Error: Not found response. Tag not compatible with AID F222222222.
    at ACR122Reader.handle_Iso_14443_4_Tag (/Volumes/CaseSensitive/nfc/node_modules/nfc-pcsc/dist/Reader.js:598:21)

ACS ACR122U PICC Interface CARD_REMOVED { atr: <Buffer 3b 86 80 01 43 68 69 70 31 37 33>,
  standard: 'TAG_ISO_14443_4',
  type: 'TAG_ISO_14443_4' }

```

While the white card, works as expected

```
ACS ACR122U PICC Interface CARD_INSERTED { atr:
   <Buffer 3b 8f 80 01 80 4f 0c a0 00 00 03 06 03 00 01 00 00 00 00 6a>,
  standard: 'TAG_ISO_14443_3',
  type: 'TAG_ISO_14443_3',
  uid: 'a7337062' }

```

zacharycohn, 2020-06-02:
Can you link me to where you got the ones that work?

Xample, 2020-06-03:
@zacharycohn sure, I bought those: https://www.velleman.eu/products/view/?id=438176 but think the problem does not comes with the regular cards, it does with the protocol or those cards. The same reader is working and reading UID properly using @Sennar4 link. I therefore replaced `nfc-pcsc` with `simple-pcsc1`, as the bundle `simple-pcsc1`. I also rewrote them in typescript. The code can be improved, but I am using this for a POC…

Reader.ts
```typescript
'use strict';

const pcsclite = require('pcsclite');
const EventEmitter = require('events');
const Reader = require('./Reader');

export class NFC extends EventEmitter {

	pcsc = null;
	logger = null;

	constructor(logger = console) {
		super();

		this.pcsc = pcsclite();

		this.logger = logger ?? this.emptyLogger();

		this.pcsc.on('reader', (reader) => {
			this.logger.info('New reader detected', reader.name);
			const device = new Reader(reader, this.logger);
			this.emit('reader', device);

		});

		this.pcsc.on('error', (err) => {
			this.logger.info('PCSC error', err.message);
			this.emit('error', err);
		});
	}

	private emptyLogger() {
		return {
			log: () => {
			},
			info: () => {
			},
			warn: () => {
			},
			error: () => {
			}
		};
	}

	close() {
		this.pcsc.close();
	}
}
```

and
NFC.ts
```typescript
'use strict';

const pcsclite = require('pcsclite');
const EventEmitter = require('events');
const Reader = require('./Reader');

export class NFC extends EventEmitter {

	pcsc = null;
	logger = null;

	constructor(logger = console) {
		super();

		this.pcsc = pcsclite();

		this.logger = logger ?? this.emptyLogger();

		this.pcsc.on('reader', (reader) => {
			this.logger.info('New reader detected', reader.name);
			const device = new Reader(reader, this.logger);
			this.emit('reader', device);

		});

		this.pcsc.on('error', (err) => {
			this.logger.info('PCSC error', err.message);
			this.emit('error', err);
		});
	}

	private emptyLogger() {
		return {
			log: () => {
			},
			info: () => {
			},
			warn: () => {
			},
			error: () => {
			}
		};
	}

	close() {
		this.pcsc.close();
	}
}

```

TommyLiang1, 2022-07-21:
Hi @pokusew,

I got the MiFare Classic 1K card to read/write with the NFC card reader and now I'm trying to test it for any credit card.

I am receiving the same error as posted in this issue.

I am testing it using ```npm run example```. I configured that to run with MiFare Classic 1K cards by authorizing the card and changing to buffer 16 spaces instead of the default of 4.

Why do you think I am receiving this? Is it because the credit cards are more secure than MiFare Classic 1K cards?
How can I get it working?

Environment Information:
OS: Ubuntu 20.04
Node Version: 16.16.0
NPM Version: 8.11.0
Card Reader: ACR122U

arslan-mert, 2023-05-18:
root@firefly:/var/nfc/credit-card-reader/src# node test_nfc.js
ACS ACR1252 Dual Reader [ACR1252 Dual Reader PICC] 00 00 device attached
ACS ACR1252 Dual Reader [ACR1252 Dual Reader SAM] 01 00 device attached
ACS ACR1252 Dual Reader [ACR1252 Dual Reader PICC] 00 00 card detected {
  atr: <Buffer 3b 8f 80 01 80 4f 0c a0 00 00 03 06 03 00 01 00 00 00 00 6a>,
  standard: 'TAG_ISO_14443_3',
  type: 'TAG_ISO_14443_3',
  uid: '876ad9ed'
}
ERROR: TransmitError: An error occurred while transmitting.
    at /var/nfc/credit-card-reader/node_modules/nfc-pcsc/dist/Reader.js:241:25 {
  code: 'failure',
  previous: [Error: SCardTransmit error: Transaction failed.(0x80100016)]
}
ACS ACR1252 Dual Reader [ACR1252 Dual Reader PICC] 00 00 card removed {
  atr: <Buffer 3b 8f 80 01 80 4f 0c a0 00 00 03 06 03 00 01 00 00 00 00 6a>,
  standard: 'TAG_ISO_14443_3',
  type: 'TAG_ISO_14443_3',
  uid: '876ad9ed'
}
hi guys
I also got these errors, I wonder how can I solve these problems https://github.com/7s4r/credit-card-reader I ran this link.My aim is to print the information of both credit and classic cards on the screen.

arslan-mert, 2023-05-18:
hi guys
I also got these errors, I wonder how can I solve these problems https://github.com/7s4r/credit-card-reader I ran this link.My aim is to print the information of both credit and classic cards on the screen.
node test_nfc.js source code 
import nfcPcsc from "nfc-pcsc";
import tlv from "node-tlv";
import luhn from "fast-luhn";
import creditCardType from "credit-card-type";

const nfc = new nfcPcsc.NFC();

nfc.on("reader", (reader) => {
    console.info(`${reader.reader.name} device attached`);

    reader.autoProcessing = true;
    reader.aid = "A0000000031010"; // EMV AID değeri
    

    reader.on("card", async (card) => {
        console.info(`${reader.reader.name} card detected`, card);

        // EMV kart verilerini almak için GPO (Get Processing Options) komutunu gönderme
        const gpoCommand = Buffer.from([0x80, 0xA8, 0x00, 0x00, 0x02, 0x83, 0x00, 0x00]);
        const gpoResponse = await reader.transmit(gpoCommand, 256);
        console.info("GPO Response:", gpoResponse.toString("hex"));

        // GPO cevabından AFL (Application File Locator) verisini alın
        const afl = gpoResponse.slice(gpoResponse.length - 4);
        console.info("AFL:", afl.toString("hex"));

        // AFL'deki her bir uygulama için kart verilerini alın
        const applications = parseAFL(afl);

        for (const application of applications) {
            // Kartın uygulama verilerini almak için READ RECORD komutunu gönderin
            const readRecordCommand = Buffer.from([0x00, 0xB2, application.sfi << 3 | 0x04, application.recordNumber, 0x00]);
            const readRecordResponse = await reader.transmit(readRecordCommand, 256);
            console.info("READ RECORD Response:", readRecordResponse.toString("hex"));

            // Kart verilerini TLV formatında analiz edin
            const tags = tlv.parse(readRecordResponse.toString("hex"));
            console.log(tags);

            const cardNumber = tags.find("5A");
            const cardExpiryDate = tags.find("5F24");

            if (cardNumber) {
                const cardNumberValue = cardNumber.value;
                const cardType = creditCardType(cardNumberValue);

                if (cardType && cardType.length > 0) {
                    console.info("Card type:", cardType[0].niceType);
                }

                console.info("Card number:", cardNumberValue);
                console.info("Card expiry date (YYMMDD):", cardExpiryDate && cardExpiryDate.value);
                console.info("Is valid:", luhn(cardNumberValue));
            } else {
                console.error("Card number not found!");
            }
        }

        reader.close();
        nfc.close();
    });

    reader.on("card.off", (card) => {
        console.info(`${reader.reader.name} card removed`, card);
    });

    reader.on("error", (err) => {
        console.error(`${reader.reader.name} an error occurred`, err);
    });

    reader.on("end", () => {
        console.info(`${reader.reader.name} device removed`);
    });
});

nfc.on("error", (err) => {
    console.error("An error occurred:", err);
});

function parseAFL(afl) {
    const applications = [];
    let offset = 0;

    while (offset < afl.length) {
        const sfi = afl[offset] >> 3;
        const startRecord = afl[offset + 1];
        const endRecord = afl[offset + 2];
        const recordNumber = startRecord;

        for (let i = startRecord; i <= endRecord; i++) {
            applications.push({ sfi, recordNumber: i });
        }

        offset += 4;
    }

    return applications;
}
compile output ;
root@firefly:/var/nfc/credit-card-reader/src# node test_nfc.js
ACS ACR1252 Dual Reader [ACR1252 Dual Reader PICC] 00 00 device attached
ACS ACR1252 Dual Reader [ACR1252 Dual Reader SAM] 01 00 device attached
ACS ACR1252 Dual Reader [ACR1252 Dual Reader PICC] 00 00 an error occurred Error: Response status error.
    at ACR122Reader.handle_Iso_14443_4_Tag (/var/nfc/credit-card-reader/node_modules/nfc-pcsc/dist/Reader.js:613:21)
ACS ACR1252 Dual Reader [ACR1252 Dual Reader PICC] 00 00 card removed {
  atr: <Buffer 3b 8a 80 01 86 65 00 a7 08 c0 55 00 90 00 42>,
  standard: 'TAG_ISO_14443_4',
  type: 'TAG_ISO_14443_4'
}
ACS ACR1252 Dual Reader [ACR1252 Dual Reader PICC] 00 00 card detected {
  atr: <Buffer 3b 8f 80 01 80 4f 0c a0 00 00 03 06 03 00 01 00 00 00 00 6a>,
  standard: 'TAG_ISO_14443_3',
  type: 'TAG_ISO_14443_3',
  uid: '876ad9ed'
}
(node:2721) UnhandledPromiseRejectionWarning: TransmitError: An error occurred while transmitting.
    at /var/nfc/credit-card-reader/node_modules/nfc-pcsc/dist/Reader.js:241:25
(Use `node --trace-warnings ...` to show where the warning was created)
(node:2721) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:2721) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
ACS ACR1252 Dual Reader [ACR1252 Dual Reader PICC] 00 00 card removed {
  atr: <Buffer 3b 8f 80 01 80 4f 0c a0 00 00 03 06 03 00 01 00 00 00 00 6a>,
  standard: 'TAG_ISO_14443_3',
  type: 'TAG_ISO_14443_3',
  uid: '876ad9ed'
}
nfc reader ---> acr1252u ,   the board i work on ---> firefly roc-rk3566 arm64 , ubuntu 20.04 (lts)