pokusew/nfc-pcsc #98: Clear Tag, Write URL and set Write-Lock

SvenTiigi, 2020-04-22:
Hey @pokusew,

First of all thanks for this awesome library! 👍 

I'm writing a small Node app and using your library to perform the following task when a NFC Tag has been detected:

- Parse header to check if the Tag has write permissions
- Clear any previously stored data
- Write an URL record
- Setting a Write-Lock / Read-Only

For parsing and constructing the NDEF record I am using [nfccard-tool](https://github.com/somq/nfccard-tool)

Currently my code looks like this:

```javascript
const { NFC } = require('nfc-pcsc');
const nfcCard = require('nfccard-tool');

...

reader.on('card', card => {
    reader
        .read(0, 20)
        .then((cardHeader) => {
            return nfcCard
                .parseInfo(cardHeader);
        })
        .then(cardInfo => {
            // Check if a write permissions are not available
            if (!cardInfo.parsedHeader.hasWritePermissions) {
                return Promise.reject('Write Permission disabled');
            }
        })
        .then(_ => {
            // Clear any previously stored Data
            return reader
                .write(4, nfcCard.prepareBytesToWrite([{}]).preparedData);
        })
        .then(_ => {
            // Initialize Data that should be written
            const data = nfcCard.prepareBytesToWrite([
                {
                    type: 'uri',
                    uri: 'https://github.com'
                }
            ]);
            // Write Data
            return reader
                .write(4, data.preparedData);
        })
        .then(() => {
            console.log('Set a Write-Lock...?')
        })
        .then(_ => console.log('Finished'))
        .catch(error => console.error('Failure', error));
});
```

I would be really grateful if you have any idea how to set a write lock or in other words make the NFC Tag `read-only` and if clearing the previously stored data is correct.

Environment-Info:
OS: macOS 10.15.4
Node-Version: v13.13.0
NPM-Version: 6.14.4
NFC-Reader: ACR122U

closed 2020-04-22 by SvenTiigi

tobsn, 2020-06-04:
hey @SvenTiigi did you ever figure out how to write that lock bit? :)

SvenTiigi, 2020-06-04:
Hey @tobsn,

In my case this worked fine:

```javascript
reader.write(3, Buffer.from([0x0, 0x0, 0x0, 0x0F]));
```

risurina, 2023-02-26:
@SvenTiigi 
did you try that in Mifare Classic?