pokusew/nfc-pcsc #101: Help with SPI and I2C connected PN532 reader

ronie19, 2020-08-02:
Is this compatible with PN532 reader connected by SPI/I2C ? 
Or only USB devices ?

PTS93, 2020-09-23:
You would have to implement the PCSC API which this package is using to talk to NFC readers. This is unlikely to be worth it, you will wanna directly talk to your reader.

ronie19, 2020-09-23:
> You would have to implement the PCSC API which this package is using to talk to NFC readers. This is unlikely to be worth it, you will wanna directly talk to your reader.

Any possible ways of doing that ? I am trying to get it recognised by pcsc_scan command in terminal. But no way I am able to get it via spi. I am only able to operate it with libnfc.

PTS93, 2020-09-23:
To be fair, if you are unsure on how to approach that the task of implementing a pcsc driver is probably too much.
libnfc already allows you to interact with the NFC reader and access all its functionality. I would suggest you use that to integrate with your application.

jonasclaes, 2022-01-11:
ifdnfc is your solution

ronie19, 2022-01-13:
> ifdnfc is your solution

Yes, I did it. However it worked with 64-bit linux due to buffer length issues in 32-bit linux