pokusew/nfc-pcsc #114: Electronjs, preload.js, NON context aware native modules not allowed

DarDoro, 2021-08-04:
I'm using module to create dedicated client with WEB NFC compliance/emulator for desktop OS (WEB NFC currently works on mobile only). 

But this module cannot be used in `preload.js` of electron app https://github.com/electron/electron/issues/18397
It can work with `app.allowRendererProcessReuse = false;`, but it will be disabled in new version.

> ```
> Timeline
> Land the app.allowRendererProcessReuse option in Electron 6
> Have the first deprecation warning land in Electron 7 for non-context aware native modules
> Have a deprecation warning land in Electron 8 for the default value of app.allowRendererProcessReuse to switch
> Switch the default value of app.allowRendererProcessReuse to true in Electron 9
> Deprecate the ability to change app.allowRendererProcessReuse in Electron 10
> Remove the ability to change app.allowRendererProcessReuse in Electron 14
> 
> ```

> VM263 renderer_init.js:93 Error: Loading non-context-aware native module in renderer: '\\?\c:\work\htnfc\node_modules\@pokusew\pcsclite\build\Release\pcsclite.node', but app.allowRendererProcessReuse is true. See https://github.com/electron/electron/issues/18397.
>     at process.func [as dlopen] (VM258 asar_bundle.js:5)
>     at Object.Module._extensions..node (VM231 loader.js:1138)
>     at Object.func [as .node] (VM258 asar_bundle.js:5)
>     at Module.load (VM231 loader.js:935)
>     at Module._load (VM231 loader.js:776)
>     at Function.f._load (VM258 asar_bundle.js:5)
>     at Function.o._load (VM263 renderer_init.js:33)
>     at Module.require (VM231 loader.js:959)
>     at require (VM238 helpers.js:88)
>     at bindings (VM92 c:\work\htnfc\node_modules\bindings\bindings.js:112)
>

alba-ado, 2021-12-16:
Hello, 

I've notified the pcsclite team about this and they have applied a fix to it, in https://github.com/santigimeno/node-pcsclite/pull/96
Is there any way this could be incorporated into your library?

Thanks in advance.

alba-ado, 2021-12-16:
I have confirmed that santigimeno/node-pcsclite#96 is indeed working with this library in electron 16.0.4

It would be great if you can fix this and publish it again. Also, why do you use your own library as a dependency instead of the original one?

alba-ado, 2022-01-20:
Is there any update on this???