pokusew/nfc-pcsc #116: Detect Readeer

lottafederico, 2021-09-29:


atoikka, 2022-03-28:
@lottafederico by any chance, did you find out anything?
I am currently seeing the same issue periodically. The next time I encounter, I'll try to look at pcsc_scan and see what it reports.

mirkogeest, 2022-05-28:
I have exactly the same issue, but with pi 3. Did you manage to find a solution to this?

atoikka, 2022-05-28:
I found out that it is likely a ACR122U reader firmware, I was able to find some references such as [this](https://social.technet.microsoft.com/Forums/en-US/20eadf9f-47c6-41a1-b4c8-d622706da1af/acr122u-smart-card-reader-not-recognized-after-reboot?forum=posready) and verified from the pcscd logs that it was unable to enumerate the device unless a complete power reset (i.e. replug of RPi power or the ACR122U itself) was done. I also experimented running pcscd on the foreground (with verbose logging) and issuing [USB reset](https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line) to the device - the results were odd, the ACR122U was functioning after about 20% of all resets.
We briefly discussed the idea of using a GPIO pin to drive the device's power supply, but found a good delivery option for the ACR1252U. I can confirm that the ACR1252U survives USB resets (no power loss) properly, and advise towards using it.

mirkogeest, 2022-05-28:
For our topology we finally found a solution, as it is not an option to unplug the usb physically.
We use a balena docker image, so we added the uhubctl app, which allows turn usb off and on, to the Dockerfile.template:

`RUN install_packages pcscd libusb-dev libpcsclite1 libpcsclite-dev dh-autoreconf python3 uhubctl g++`

And at package.json we start this way:
```
  "scripts": {
    "start": "uhubctl -l 1-1 -p 2 -a off && uhubctl -l 1-1 -p 2 -a on && pcscd && node server.js"
  }
```

It's working like a charm!

ltcmdrkeen, 2024-01-05:
Using a different driver solved this issue for me:

## Issue
- the native driver does not initialize correctly. we need to manually disconnect and reconnect the USB device.
- Using another driver seems to work: https://www.acs.com.hk/en/driver/3/acr122u-usb-nfc-reader/
- as there is no ARM64 build / release currently, we need to manually compile it from the sources at from https://github.com/acshk/acsccid
- the latest version (1.1.10) is corrupt (missing m4 files), but an earlier version works: 1.1.8 (see nfc-driver/acsccid-1.1.8.zip in this repo)

## Build acsccid driver:
- use version 1.1.8, newer versions are broken due to missing m4 folder 
- install dependencies: `sudo apt-get install libtool libusb-1.0-0-dev`
```bash
tar -zxvf acsccid-1.1.8.zip
- cd acsccid-1.1.8/
- ./bootstrap
- ./configure
- make
- make install
```

mirkogeest, 2024-10-21:
> @mirkogeest , would love to see your full configuration for this. I'm having a devil of a time getting my ACR122U to work with this library in balena.
> 
> > For our topology we finally found a solution, as it is not an option to unplug the usb physically. We use a balena docker image, so we added the uhubctl app, which allows turn usb off and on, to the Dockerfile.template:
> > `RUN install_packages pcscd libusb-dev libpcsclite1 libpcsclite-dev dh-autoreconf python3 uhubctl g++`
> > And at package.json we start this way:
> > ```
> >   "scripts": {
> >     "start": "uhubctl -l 1-1 -p 2 -a off && uhubctl -l 1-1 -p 2 -a on && pcscd && node server.js"
> >   }
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > It's working like a charm!

Hi Trankin,

This is Dockerfile.template:
```
FROM balenalib/%%BALENA_MACHINE_NAME%%-node:17.7.2-sid-build

RUN install_packages libglib2.0-dev

# Now install the packages we need:
RUN install_packages pcscd libusb-dev libpcsclite1 libpcsclite-dev dh-autoreconf python3 uhubctl g++

# Defines our working directory in container
WORKDIR /usr/src/app

# Copies the package.json first for better cache on later pushes
COPY package*.json ./

RUN JOBS=MAX npm install --production --unsafe-perm && npm cache verify && rm -rf /tmp/*

# This will copy all files in our root to the working directory in the container
COPY . ./

# Enable udevd so that plugged dynamic hardware devices show up in our container.
ENV UDEV=1

# server.js will run when container starts up on the device
CMD ["npm", "start"]
```

And here the package.json:
```
{
  "version": "0.0.4",
  "main": "index.js",
  "license": "MIT",
  "private": true,
  "type": "module",
  "scripts": {
    "deploy:staging": "balena push 1663111.local",
    "deploy:production:leds:8": "balena push xxxxx/yyyyy",
    "start": "uhubctl -l 1-1 -p 2 -a off && uhubctl -l 1-1 -p 2 -a on && pcscd && node server.js"
  },
  "dependencies": {
    "@gbkwiatt/node-rpi-ws281x-native": "^1.0.1",
    "axios": "^1.3.6",
    "concurrently": "^8.0.1",
    "dayjs": "^1.11.7",
    "dotenv": "^16.0.3",
    "jsonwebtoken": "^9.0.0",
    "mongodb": "^5.3.0",
    "nfc-pcsc": "^0.8.1",
    "node-fetch": "^3.3.1",
    "sqlite3": "^5.1.6"
  },
  "devDependencies": {
    "env-cmd": "^10.1.0"
  }
}
```

I hope this helps you!