pokusew/nfc-pcsc #118: Can I use it in chrome extension?

bryht, 2021-11-02:
Anyone knowing about the question? want to add NFC read ability to my website.   Or anyone has other ideas? Thanks in advance.

pokusew, 2021-11-02:
Hi @bryht,

unfortunately, nfc-pcsc **cannot** be used in Chrome extensions. 😢 nfc-pcsc can be **only in Node.js environments** because it uses Node Native Modules to access system PC/SC API. So the Node.js runtime and PC/SC API are required for nfc-pcsc to run. That makes it possible to use it on most OS (Windows, macOS, Linux) directly in [Node.js](https://nodejs.org/en/) or in [Electron.js](https://www.electronjs.org/) and desktop apps.

Depending on your use case, there are different possibilities for the use of NFC directly in the browsers (whether directly in web pages/apps or in browser extensions):
* Web NFC API – experimental web standard, only works in Chrome 89+ on Android, works with built-in only supports NDEF formatted tags, see [web.dev/nfc](https://web.dev/nfc/).
* Web USB API – experimental web standard, [usable](https://caniuse.com/webusb), allows accessing USB devices. You can implement support for your USB NFC reader. Try [this fork of chrome-nfc](https://github.com/beaufortfrancois/chrome-nfc), which uses Web USB and supports quite common reader ACR122U.
* Web HID API, Web Serial API – depending on your reader, you could implement communication using these experimental web standards.
* browser web app communicating via [WebSocket](https://developer.mozilla.org/en-US/docs/Web/API/WebSockets_API) with a Node.js server (on localhost) running [nfc-pcsc](https://github.com/pokusew/nfc-pcsc) – that's perfectly doable, and you have complete control over the NFC implementation if you are okay with running a background Node.js app. But maybe, in that case, you could 
ship your product directly as a desktop app based on [Electron.js](https://www.electronjs.org/).

Hope this answers your question. 🙂 Let me know if you need anything else.

bryht, 2021-11-03:
Hi @pokusew,

Thank you for the detail answers, they are really helpful, follow your advice, I have tried Web HID API and Web Serial API in chrome console:
`await navigator.hid.requestDevice({filters:[]})`
`await navigator.serial.requestPort()`

 unfortunately they cannot list my ACR122U, 
[image]
 but I will try more, hopefully will get some result.

closed 2021-11-03 by bryht