pokusew/nfc-pcsc #119: Writing in sectors other than 4, 5 or 6 appears to literally break the NFC tag's memory

nSneerfulBike, 2021-11-14:
Literally what I said in the title, after writing to those sectors on my MIFARE Classic 1K I can't do anything else with my NFC tags. What's happening and how can I fix that?

pokusew, 2021-11-14:
Hi @nSneerfulBike,

You are probably writing application data into so-called _sector trailers_.

Take a look into [MIFARE Classic 1K data sheet](https://www.nxp.com/docs/en/data-sheet/MF1S50YYX_V1.pdf). The Memory organization is described in Section 8.6:
> ... memory is organized in 16 sectors of 4 blocks. One block
contains 16 bytes.

**What's important**, it's that the last block in each sector is the "sector trailer" (see Section 8.6.3), which contains **access conditions** for the other 3 data blocks of the sector.

In other words, sector trailers are not for application data. If you write some random data to it, you will change access conditions and authentication keys A and B for the corresponding sector in an unexpected way.

When accessing these blocks with this library, blocks are numbered as follows:
- sector 0:
  - block 0 – manufacturer data (read only)
  - block 1 – data block
  - block 2 – data block
  - **block 3** – sector trailer
- sector 1:
  - block 4 – data block
  - block 5 – data block
  - block 6 – data block
  - **block 7** – sector trailer
- sector 2:
  - block 8 – data block
  - block 9 – data block
  - block 10 – data block
  - **block 11** – sector trailer
- and so on

I hope it explains and solves your problem. 🙂 I've closed the issue, but feel free to comment on it or re-open if you have further questions or face any problems.

---

PS Don't forget to star ⭐️ my library if you find it useful. 😃 Thanks.

closed 2021-11-14 by pokusew

nSneerfulBike, 2021-11-14:
I get how it works, I solved the problem now after making a lot of attempts and destroying so many sectors. I'd really consider this a bug, since if I have let's say a 64-bytes long Buffer and I want to write it, I'd assume that this module takes care of not destroying the sectors, but it doesn't. It just writes everything and doesn't care about the sector trailers.