pokusew/nfc-pcsc #120: Desfire 4k EV1 reading UID getting 91 1c response

RobertRozic, 2021-11-19:
Hello,

I am trying to obtain UID of desfire 4K EV1 card using this library. 
I am using ACR122U card reader and I installed latest available drivers.

My code is working on Linux but when I rebuild it on Windows I am getting error response after last APDU command. 

I am sending 3 APDU commands
1. 90 60 00 00 00
I receive correct response ending with 91 AF which means that additional data frame is expected to be sent. 

Next command is 
2. 90 AF 00 00 00
Where I also receive response ending with 91 AF

After I send last command
3. 90 AF 00 00 00
This is where it fails on windows.
On Linux I receive response where I can read cards UID (7 first bytes) but on windows I receive 91 1C response.

Did anybody experienced same behaviour and managed to fix this issue?

Kind regards

cyrstron, 2024-06-11:
I am having the same problem specifically on Windows. For now, it seemed like setTimeout helps, but that is a very unsettling solution.

cyrstron, 2024-06-12:
So it seems that WIN sends some commands to a card under the hood and breaks the transaction. 

I used `pcsclite-node` with exclusive `share_mode` and it helped. But it seems that in this wrapper you can't set it.