pokusew/nfc-pcsc #122: Latest releases not on NPM?

PanMan, 2021-12-08:
On NPM the latest release is `0.6.0`: https://www.npmjs.com/package/@pokusew/pcsclite 
Here it's `0.8.1`: https://github.com/pokusew/nfc-pcsc/releases/tag/v0.8.1

Is there a reason the later releases aren't published on NPM?

pokusew, 2021-12-08:
Hi @PanMan,

all releases have always been published both on GitHub and on npm. 📦

**But** there are **two different** packages:
1. **nfc-pcsc** [npm 0.8.1](https://www.npmjs.com/package/nfc-pcsc), [GitHub 0.8.1](https://github.com/pokusew/nfc-pcsc) – High-level easy-to-use NFC library, it contains implementation of common PC/SC commands to support reading/writing NFC card and tags, it uses @pokusew/node-pcsclite under the hood.
1. **@pokusew/pcsclite** [npm 0.6.0](https://www.npmjs.com/package/@pokusew/pcsclite), [GitHub 0.6.0](https://github.com/pokusew/node-pcsclite) – Lower-level library, virtually it's just bindings (Node Native Module) over pcsclite (which is an API provided by common OS, incl. macOS, Linux, Windows) to access Smart Cards (incl. NFC).

I hope this clears things up. 🙂 Let me know if you have any further questions.

closed 2021-12-08 by pokusew

PanMan, 2021-12-08:
Thanks a lot and sorry for the confusion!!